forkineye/ESPixelStick #664: xLights FPP Connect on Ethernet fails - DigQuad

bmasella, 2023-09-17:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-ci5943922254

**Hardware Version**
DigOcta

**Binary release or compiled yourself?**

**Operating System (and ver
Windows 11

**Web Browser (and version)**
Chrome

**Access Point**

**Describe the bug**
xLights version 2023.13 and 2023.14 FPP Connect seems to connect to the Ethernet and starts the send process but it is extremely slow... eventually xLights will crash...  and not all files are in the file manager area.  Tried doing wireless instead and FPP Connect in xLights doesn't find the wireless instance of the DigOcta.   Upload of the OUTPUTS works just fine and they are configured correctly via Ethernet so I know that there is an ethernet connection between xLights and the DigOcta... Seems to be an issue with the network stuff on DigOcta?

bmasella, 2023-09-17:
Also for those that do make it over before xLights crashes there is a strange timestamp on them.

[image]

MartinMueller2003, 2023-09-17:
As I said, the date is going to be some relatively random value unless xLights sets the date on the ESP prior to sending the files. Try this

- Turn off the browser going to the ESP.
- Reboot the ESP. It now has some random starting date in it.
- Use FPP connect to upload a few files
- Bring up the browser and note the unexpected dates. 
- FYI: Bringing up the browser, causes the ESP date to get set to the browsers date/time.
- Now have FPP Connect upload a few more files to the ESP
- Look at the date / time on the files. It should be current (not exact but within a few minutes.

You should create an xLights request to add setting the time on the ESP before uploading any files.

On the flip side, there have been requests by people to have the ESP act as an FPP master. That means we may need to add NTP to the ESP which would allow the device to get the current time on its own. (if an NTP server is reachable). Most local routers also support NTP server support.

MartinMueller2003, 2023-09-17:
Dup of #655

closed 2023-12-08 by MartinMueller2003