forkineye/ESPixelStick #667: SD Card not found after flashing

oneshot678, 2023-09-22:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
ESPixelStick v4.0-ci6236846991 (Sep 19 2023 - 14:19:57)

**Hardware Version**
Quinled Quad AE+ 8 Pixel Port

**Binary release or compiled yourself?**
ESPixelStick v4.0-ci6236846991 (Sep 19 2023 - 14:19:57)

**Operating System (and version)**
Windows 10 64X

**Web Browser (and version)**
Chrome Version 117.0.5938.92 (Official Build) (64-bit)

**Access Point**
no Home WiFi

**Describe the bug**
23:04:37: [  2433][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
23:04:38: [FileMgr] No SD card installed

Tried multiple 32GB SD cards, and 2 separate esp32/hats but the file manager tab never shows up.

MartinMueller2003, 2023-09-22:
Turn on advanced mode and look at the SD card GPIO settings. Check to see if they match those on the base card.

oneshot678, 2023-09-22:
Here are the settings that are currently set up
from the advance device setup page
SD MISO Pin: 19
SD MOSI Pin: 23
SD Clock Pin: 18
SD Chip Select Pin: 5

[image: image.png][image: image.png]

MartinMueller2003, 2023-09-22:
OK Those match the numbers set in the SW Defaults. You said you have tried multiple AE+ cards and none of them work? When WLED was installed, could it see the SD card?

oneshot678, 2023-09-22:
I had not tried it. I did not know there was a place in WLED to look for an
SD card. But good news I pulled an SD card from my 3d Printer Sandisk High
Endurance card  32GB and the file manager came up! The cards that did not
work in the ESP32 were sandisk Ultra's  they were able to be written to in
my laptop but did not work on the esp32. I ran a check on the cards with
came back clean. I purchased all 5 cards including the high Endurance card
on amazon "SanDisk Store" which i thought would be leget. I will try to go
to a bestbuy to test if i had bad cards from amazon
.  [image: image.png]  [image: image.png]

cybercop23, 2023-09-22:
I have the same HW and used the same CLI and worked. Ron has the ESPixelStick v3 hw and he has no issues also reading and playing from the card. AFAIK there's nothing in WLED that supports the SD card.

cybercop23, 2023-09-22:
Nick was on Zoom and his new cards worked as expected. So chuck this to bad/fake cards.
I'd think we can close this issue now.

closed 2023-09-22 by MartinMueller2003

thughes2001, 2024-01-17:
Having same problem. When flashing, Firmware can read all file names on SD card then says card not installed. Using Quinled Uno AE+ software.
Below is from the install serial output.

22:22:05: [FileMgr] SD Card Size: 7580MB
22:22:08: [FileMgr] > System Volume Information
22:22:08: [FileMgr]   WPSettings.dat - 12
22:22:08: [FileMgr]   IndexerVolumeGuid - 76
22:22:08: [FileMgr] 12 Days.fseq - 3031537
22:22:08: [FileMgr] Linus Christmas Speech.fseq - 6035
22:22:08: [FileMgr] The_First_Noel.fseq - 113547
22:22:08: [FileMgr] Alvin 40fps.fseq - 953129
22:22:08: [FileMgr] Bob Seger Little Drummer boy.fseq - 40565
22:22:08: [FileMgr] Christmas Eve-Sarajevo.fseq - 709293
22:22:08: [FileMgr] Frosty.fseq - 82448
22:22:08: [FileMgr] Here Comes Santa.fseq - 263662
22:22:08: [FileMgr] It is begining to look like Christmas.fseq - 253033
22:22:08: [FileMgr] Configuration File: '/config.json' loaded.
22:22:08: [  6480][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
22:22:11: [  6980][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
22:22:11: [FileMgr] No SD card installed

thughes2001, 2024-01-17:
Connect to module with XLights and sees everything but the SD card. Acts like there is no SD Card so no file manager shows up.Can upload outputs from XLights. Even tried running an Ardinio sketch to test the SD card and worked fine.
Tim Hughes

 
 

What happens after a power cycle?

thughes2001, 2024-01-17:
I see the error is created in sd_disio.cpp. I was looking for that scr file to look and see what it is missing but do not see that file anywhere in your Github. Wanted to look at the source code to see how it generates that error.
Tim Hughes

thughes2001, 2024-01-17:
Here is the complete serial output from flashing the module

08:09:01: [   ESP] ESPixelStick v4.0-beta5 (Dec  1 2023 - 17:11:34)
08:09:01: [   ESP] v4.4.1-1-gb8050b365e
08:09:01: [FileMgr] Flash file system initialized. Used = 221184 out of 327680
08:09:01: [FileMgr] Listing directory: /
08:09:01: [FileMgr] 'UpdRecipe.json':     '784'
08:09:01: [FileMgr] 'config.json':     '620'
08:09:01: [FileMgr] 'input_config.json':     '1980'
08:09:01: [FileMgr] 'output_config.json':     '8308'
08:09:01: [FileMgr] Listing directory: //www
08:09:01: [FileMgr] 'alexa.html.gz':     '147'
08:09:02: [FileMgr] 'apa102.html.gz':     '1494'
08:09:02: [FileMgr] 'artnet.html.gz':     '384'
08:09:02: [FileMgr] 'ddp.html.gz':     '145'
08:09:02: [FileMgr] 'dmx.html.gz':     '454'
08:09:02: [FileMgr] 'e1_31.html.gz':     '462'
08:09:02: [FileMgr] 'effects.html.gz':     '971'
08:09:02: [FileMgr] 'esps.css.gz':     '13393'
08:09:02: [FileMgr] 'esps.js.gz':     '60981'
08:09:02: [FileMgr] 'favicon.ico':     '6450'
08:09:02: [FileMgr] 'fpp_remote.html.gz':     '346'
08:09:02: [FileMgr] 'gece.html.gz':     '499'
08:09:02: [FileMgr] 'gs8208.html.gz':     '1575'
08:09:02: [FileMgr] 'index.html.gz':     '4937'
08:09:02: [FileMgr] 'mqtt.html.gz':     '648'
08:09:02: [FileMgr] 'relay.html.gz':     '505'
08:09:02: [FileMgr] 'renard.html.gz':     '528'
08:09:02: [FileMgr] 'serial.html.gz':     '619'
08:09:02: [FileMgr] 'servo_pca9685.html.gz':     '519'
08:09:02: [FileMgr] 'tls3001.html.gz':     '1495'
08:09:02: [FileMgr] 'tm1814.html.gz':     '1610'
08:09:02: [FileMgr] 'ucs1903.html.gz':     '1576'
08:09:02: [FileMgr] 'ucs8903.html.gz':     '1578'
08:09:02: [FileMgr] 'ws2801.html.gz':     '1492'
08:09:02: [FileMgr] 'ws2811.html.gz':     '1577'
08:09:02: [FileMgr] SD Card Size: 7580MB
08:09:02: [FileMgr] > System Volume Information
08:09:02: [FileMgr]   WPSettings.dat - 12
08:09:02: [FileMgr]   IndexerVolumeGuid - 76
08:09:02: [FileMgr] test.txt - 1048576
08:09:02: [FileMgr] foo.txt - 13
08:09:02: [FileMgr] 12 Days.fseq - 3031537
08:09:02: [FileMgr] Linus Christmas Speech.fseq - 6035
08:09:02: [FileMgr] The_First_Noel.fseq - 113547
08:09:02: [FileMgr] Alvin 40fps.fseq - 953129
08:09:02: [FileMgr] Bob Seger Little Drummer boy.fseq - 40565
08:09:02: [FileMgr] Christmas Eve-Sarajevo.fseq - 709293
08:09:02: [FileMgr] Frosty.fseq - 82448
08:09:02: [FileMgr] Here Comes Santa.fseq - 263662
08:09:02: [FileMgr] It is begining to look like Christmas.fseq - 253033
08:09:02: [FileMgr] Configuration File: '/config.json' loaded.
08:09:02: [  3653][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
08:09:05: [  4153][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
08:09:05: [FileMgr] No SD card installed
08:09:05: [FileMgr] Configuration File: '/input_config.json' loaded.
08:09:05: [FileMgr] Configuration File: '/output_config.json' loaded.
08:09:06: [FileMgr] Configuration File: '/admininfo.json' saved 176 bytes.
08:09:06: [FileMgr] Configuration File: '/config.json' saved 387 bytes.
08:09:06: [WiFiDrv] WiFi Entering State: Connecting Using Config Credentials
08:09:07: [WiFiDrv] Using Static IP
08:09:07: [WiFiDrv] Connecting to 'shadow' as esps-94135BF9D108
08:09:07: [WiFiDrv] WiFi Entering State: Connected To AP
08:09:08: [WiFiDrv] Connected with IP: 192.168.0.206
08:09:08: [WebMgr] Web server listening on port 80
08:09:08: [  FPPD] FPPDiscovery subscribed to broadcast messages on port: 32320
08:09:08: [  FPPD] FPPDiscovery subscribed to multicast: 239.70.80.80:32320

thughes2001, 2024-01-17:
This what I get when opening from XLights
Tim Hughes

MartinMueller2003, 2024-01-17:
There is a known problem with many of the SD Card interfaces that put the SD Card into a locked up state if it is in the middle of an operation. A power cycle will clear the condition. When I design an SD interface I include a power control for this case. However, in this case it looks like there is an error reading the SD card that is causing the low level driver to have an issue. 

You mentioned that you are looking for the sd_diskio.c file. That is part of the precompiled kernel we get from espressif. Very difficult to debug issues in that space.

thughes2001, 2024-01-17:
Thanks. Still Confused about not reading the card. It reads the volume, size and files already on the card then has the problem. Could it be if the module is not setup for a software reset just before mounting, and causing the issue? As for the sd_diskio, was just curious what is driving the two error messages. Would help if it gave a small hint other than not working right after reading all the information on the SD card. Were is the low level driver located?

MartinMueller2003, 2024-01-17:
I would suspect there is a data error coming from the SD that the ESP does not know how to handle. I have a variety of SD cards. One even has 70 files on it and it works fine.

thughes2001, 2024-01-17:
Found a few more issues. Load WLED on the module and it works fine with XLights. Load ESPixelSticks and can upload to it from XLights but does not work. The only thing I have to change in X:ights is the controller Model and variant. Not sure what it is.

cybercop23, 2024-01-17:
WLED doesn't use the card, and neither does the vanila flavor of espixel stick. at least the base functionality should work. Have you tried a different card? Hop on Zoom, xLights -> Help -> Zoom and we can take a look to see what we can do for you.

thughes2001, 2024-01-18:
I first tried flashing it with ESPixel stick as a Quin UNO AE+.That is suppose utilize the SD Card that WLED can't. Even flashing the ESP32 as a plain Quin Dig Uno, It will not work with XLights. It will connect and It will take an upload from XLights and configure the outputs but when running the sequence the lights don't work. Installing WLED on the same module and reconfigure XLights for WELD Quni Uno, the lights work perfect.

MartinMueller2003, 2024-01-20:
With ESPixelStick installed and configured by xLights. Put input 2 into effects mode. Set it to rainbow. Do you get anything on the outputs? I suspect xLights is sending the wrong data to the ESP. 
Once you have verified the outputs work using effects, then try to play again from xLights. Go to the home page, what are the stats telling you? 

Include a screenshot of the home page here.

thughes2001, 2024-01-20:
With the new firmware that you directed me to. It is working. Had a hard time with XLights and the IP addresses but seem to get it worked out. I was able to get it to work as an FPP remote. Looks good so far.

Thanks Very muchTim Hughes