forkineye/ESPixelStick #669: Erratic random flickering

broomy111, 2023-09-29:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version** 4.0ci6236846991

**Hardware Version** ESP8266     ESPSV3

**Binary release or compiled yourself?** Binary

**Operating System (and version)** Windows 11

**Web Browser (and version)** Opera GX

**Access Point**

**Describe the bug**  Im running a Fseq file on the pixel stick and after it repeats twice, I start to get rand flickering on my light string towards the beginning of the strand..  I am only running 79 pixels attached to the ESP.  I've tried several versions of firmware and same results.  Tried different pixels , different 3 core cables,  created new sequence, and I see see the same result.  Any help would be appreciated.

Thank you

MartinMueller2003, 2023-09-29:
please post a screen shot of the admin page

broomy111, 2023-09-29:
Thank you Martin, here you are

[image: image.png]

MartinMueller2003, 2023-09-29:
I cant seem to open the image

broomy111, 2023-09-29:
Let me attach the pic and lets see if this works.
[image]

MartinMueller2003, 2023-10-01:
It looks like some of the sync packets are being dropped. The output driver seems fairly clean when the data is reliable. I am seeing a loss of data on the input side. More digging.

MartinMueller2003, 2023-10-02:
The only place I am seeing anything odd is at sequence start using  FPP 7. My FPP 5 is not showing the same issues. It looks like some of the start messages are just not arriving. Now I need to find out if they are being sent. Wireshark, here I come.

broomy111, 2023-10-02:
Thank you Martin for checking into it.  I'm running the FSEQ local on the
device for my neighbor's halloween display so there won't be any input
coming into the device as it wont be on a LAN.     For now I put a
beaglebones running FPP attached to a Falcon F4 and its working for now, so
its not a major rush to solve the espixel issue if you have other major
issues you are looking at.

Thanks again for the assistance

Kevin

MartinMueller2003, 2023-10-02:
It looks like the lates FPP only sends one sync message per second. That means a lost message at any point will cause a pretty hefty skip in time.

MartinMueller2003, 2023-10-31:
PR: Remote play updates #680

closed 2023-10-31 by MartinMueller2003