forkineye/ESPixelStick #670: QUINLED DIG OCTA crash/reboot with Physical Ethernet

robdmiller, 2023-10-02:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
ESPixelStick CI #525 

**Hardware Version**
QUINLED DIG OCTA

**Binary release or compiled yourself?**
Binary Release

**Operating System (and version)**
Win11 22H2

**Web Browser (and version)**
Version 117.0.5938.92

**Access Point**
Google Nest Wifi Pro

**Describe the bug**
When using physical ethernet on QUINLED DigOcta -- ESPixelStick crashes and causes reload.   Tried static and DHCP and behavior is the same, tried reflashing with a few other versions and no difference. 

`20:52:13: [FileMgr] ERROR: Cannot find 'wificonfig.json' for reading. File does not exist.
20:52:14: [FileMgr] SD file: 'wificonfig.json' not found.
20:52:14: [FileMgr] Configuration File: '/admininfo.json' saved 180 bytes.
20:52:14: [FileMgr] Configuration File: '/config.json' saved 579 bytes.
20:52:14: [WiFiDrv] WiFi Entering State: Connecting Using Config Credentials
20:52:15: [WiFiDrv] Using DHCP
20:52:15: [WiFiDrv] Connecting to 'NOSOLUTION' as ESPixelOcta
20:52:15: E (9933) wifi:Association refused temporarily, comeback time 1018 mSec
20:52:16: [WiFiDrv] WiFi Entering State: Connected To AP
20:52:17: [WiFiDrv] Connected with IP: 192.168.3.87
20:52:17: [WebMgr] Web server listening on port 80
20:52:17: [  FPPD] Listening on port 32320
20:52:17: [EthDrv] Entering State: Powering Up
20:52:24: [ 12587][E][esp32-hal-gpio.c:95] __pinMode(): Invalid pin selected
20:52:24: E (17711) gpio: gpio_set_level(226): GPIO output gpio_num error
20:52:24: [EthDrv] Entering State: Connecting
20:52:26: [EthDrv] ETH IP Before Start: 0.0.0.0
20:52:26: [EthDrv] Ethernet Connecting as ESPixelOcta
20:52:30: [ 47149][E][vfs_api.cpp:104] open(): /littlefs/www/index.html does not exist, no permits for creation
20:52:58: [266622][E][ETH.cpp:425] config(): STA IP could not be configured! Error: 20487
20:56:38: [EthDrv] Connecting to Ethernet with Static IP
20:56:38: [EthDrv] Entering State: Wait for IP Address
20:56:38: [EthDrv] Entering State: Got IP
20:56:38: [EthDrv] Ethernet Connected with IP: 10.1.1.22
20:56:38: E (320257) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
20:57:26: E (320257) task_wdt:  - async_tcp (CPU 0/1)
20:57:26: E (320257) task_wdt: Tasks currently running:
20:57:26: E (320257) task_wdt: CPU 0: IDLE
20:57:26: E (320257) task_wdt: CPU 1: loopTask
20:57:26: E (320257) task_wdt: Aborting.
20:57:26: 
20:57:26: abort() was called at PC 0x40116de4 on core 0
20:57:26: 
20:57:26: 
20:57:26: Backtrace:0x40084021:0x3ffbea9c |<-CORRUPTED
20:57:26: 
20:57:26: 
20:57:26: 
20:57:26: 
20:57:26: ELF file SHA256: 0000000000000000
20:57:26: 
20:57:26: Rebooting...
20:57:26: ets Jul 29 2019 12:21:46
20:57:26: 
20:57:26: rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:57:26: configsip: 0, SPIWP:0xee
20:57:26: clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:57:26: mode:DIO, clock div:2
20:57:26: load:0x3fff0030,len:1184
20:57:26: load:0x40078000,len:12784
20:57:26: load:0x40080400,len:3032
20:57:26: entry 0x400805e4
`

MartinMueller2003, 2023-10-05:
Please leave the CAT5 disconnected and send a screenshot of the admin page.

robdmiller, 2023-10-06:
[image]

MartinMueller2003, 2023-10-29:
I just tried this on the latest build and I am not seeing an issue. Please retest using this dist and let me know what you see: https://drive.google.com/drive/folders/1bGrkXSMoGjWtX5iWspPFRogM2G6c1VXu?usp=sharing

MartinMueller2003, 2023-11-13:
Have you reproduced this using the latest version? I would like to close this issue after you test.

robdmiller, 2023-11-14:
sorry not yet, I'll get it tested ASAP. My controller was mounted outside for Halloween show and I haven't had a chance to take it inside to flash.

apologize for the delay and thanks for all your hard work.

robdmiller, 2023-11-15:
current version seems to be working! up and running for 6 hours, no crash.  thanks again!

closed 2023-11-15 by robdmiller