forkineye/ESPixelStick #673: Issues uploading outputs from xLights.

rledonne, 2023-10-04:
I am not sure if this is an xLights bug or espixelstick.  I have 2 devices with espixelstick on them.  They both do different things with you try to upload outputs from xlights.

**ESPixelStick Firmware Version**

Device 1 - Espixelstick v3 hardware.
FW Version 4.0-beta4 Build Date Nov 13 2021 - 14:53:31 ArchitectureESP8266
Issue:  After uploading outputs, the outputs on the device setup page are disabled, and if re-enabled, it reverts to pixel count of 100, brightness 100.
[image]

Device 2 - DigQuad AE+ with 8 outputs.
FW Version 4.x-dev Build Date Sep 25 2023 - 16:50:43 Architecture ESP32 Board Name esp32_quinled_quad_ae_plus
Issue:  xLights crashes and closes when trying to upload outputs.

xLights has been updated to 2023.15, but it was doing the same thing with 2023.13.

MartinMueller2003, 2023-10-04:
The 2021 version has plenty of issues and is not a supported version. You will need to use the Sept22 or newer version of V4. 

As for xLights crashing, that needs to reported to xLights. There is nothing the ESP should be able to do that would cause such a crash.

robdmiller, 2023-10-05:
Make sure your outputs in xlights don't have gaps in them. I had crashing because I skipped some which didn't affect WLED but became an issue after flashing espixelstick.

rledonne, 2023-10-05:
> Make sure your outputs in xlights don't have gaps in them. I had crashing because I skipped some which didn't affect WLED but became an issue after flashing espixelstick.

THANK YOU!  I think that may be my issue.  I had skipped ports 5 and 6 in the xLights visualizer because I was planning on using them to add more strands to my mega tree (currently ports 1-4), and used port 7 for my star.  Sometimes it would play directly from the SD card, sometimes it wouldn't.  Sometimes it would play multisync, sometime it wouldn't.  It was driving me crazy.

I moved the star to port 5, and batch rendered and uploaded and it seems all is right with the world again.

MartinMueller2003, 2023-10-05:
Does that mean we can close this issue?

rledonne, 2023-10-05:
I think so.  I guess it raises a different issue in that things may go weird if you skip ports.  Knowing that now, it is something that I can avoid going forward.

closed 2023-10-05 by MartinMueller2003