forkineye/ESPixelStick #675: QUINLED DIG-QUAD AE+ Corrupt heap message before

DaleTiger, 2023-10-11:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-ci6397314717

**Hardware Version**
esp32_quinled_quad_ae_plus_8

**Binary release or compiled yourself?**
Binary release

**Operating System (and version)**
Microsoft Windows 11 Home

**Web Browser (and version)**
Microsoft Edge  Version 117.0.2045.60

**Access Point**
Netgear AC1450 Firmware Version V1.0.0.36_10.0.17

**Describe the bug**

SERIAL OUTPUT FROM ESPFLASHTOOL:

CORRUPT HEAP: Bad head at 0x3ffd9078. Expected 0xabba1234 got 0x3ffca17c
16:45:22: 
16:45:22: assert failed: multi_heap_free multi_heap_poisoning.c:253 (head != NULL)
16:45:22: 
16:45:22: 
16:45:22: Backtrace:0x40084089:0x3ffde1500x4008d4fd:0x3ffde170 0x400929e9:0x3ffde190 0x4009262f:0x3ffde2c0 0x40084549:0x3ffde2e0 0x40092a19:0x3ffde300 0x401709ba:0x3ffde320 0x400f3fab:0x3ffde340 0x400d229e:0x3ffde370 0x400e3787:0x3ffde3f0 0x40186891:0x3ffde410 0x401872e9:0x3ffde430 0x400f18e1:0x3ffde450 0x400f19f1:0x3ffde480 0x400f3b72:0x3ffde4d0 0x400fe43d:0x3ffde560 0x400fe496:0x3ffde600 
16:45:22: 
16:45:22: 
16:45:22: 
16:45:22: 
16:45:22: ELF file SHA256: 0000000000000000
16:45:22: 
16:45:22: Rebooting...
16:45:22: ets Jul 29 2019 12:21:46
16:45:22: 
16:45:22: rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
16:45:22: configsip: 0, SPIWP:0xee
16:45:22: clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:45:22: mode:DIO, clock div:2
16:45:22: load:0x3fff0030,len:1184
16:45:22: load:0x40078000,len:12784
16:45:22: load:0x40080400,len:3032
16:45:22: entry 0x400805e4
16:45:22: 
16:45:23: [   ESP] ESPixelStick v4.0-ci6397314717 (Oct  3 2023 - 19:05:55)
16:45:23: [   ESP] v4.4.1-1-gb8050b365e
16:45:23: [FileMgr] Flash file system initialized. Used = 217088 out of 327680
16:45:23: [FileMgr] Listing directory: /
16:45:23: [FileMgr] 'UpdRecipe.json': 	'784'
16:45:23: [FileMgr] 'admininfo.json': 	'186'
16:45:23: [FileMgr] 'config.json': 	'406'
16:45:23: [FileMgr] 'input_config.json': 	'1983'
16:45:23: [FileMgr] 'output_config.json': 	'10390'
16:45:23: [FileMgr] Listing directory: //www
16:45:23: [FileMgr] 'alexa.html.gz': 	'147'
16:45:23: [FileMgr] 'apa102.html.gz': 	'1494'
16:45:23: [FileMgr] 'artnet.html.gz': 	'384'
16:45:23: [FileMgr] 'ddp.html.gz': 	'145'
16:45:23: [FileMgr] 'dmx.html.gz': 	'454'
16:45:23: [FileMgr] 'e1_31.html.gz': 	'462'
16:45:23: [FileMgr] 'effects.html.gz': 	'971'
16:45:23: [FileMgr] 'esps.css.gz': 	'13393'
16:45:23: [FileMgr] 'esps.js.gz': 	'60967'
16:45:23: [FileMgr] 'favicon.ico': 	'6450'
16:45:23: [FileMgr] 'fpp_remote.html.gz': 	'346'
16:45:23: [FileMgr] 'gece.html.gz': 	'499'
16:45:23: [FileMgr] 'gs8208.html.gz': 	'1575'
16:45:23: [FileMgr] 'index.html.gz': 	'4939'
16:45:23: [FileMgr] 'mqtt.html.gz': 	'648'
16:45:24: [FileMgr] 'relay.html.gz': 	'505'
16:45:24: [FileMgr] 'renard.html.gz': 	'528'
16:45:24: [FileMgr] 'serial.html.gz': 	'619'
16:45:24: [FileMgr] 'servo_pca9685.html.gz': 	'519'
16:45:24: [FileMgr] 'tls3001.html.gz': 	'1495'
16:45:24: [FileMgr] 'tm1814.html.gz': 	'1610'
16:45:24: [FileMgr] 'ucs1903.html.gz': 	'1576'
16:45:24: [FileMgr] 'ucs8903.html.gz': 	'1578'
16:45:24: [FileMgr] 'ws2801.html.gz': 	'1492'
16:45:24: [FileMgr] 'ws2811.html.gz': 	'1577'
16:45:24: [FileMgr] SD Card Size: 15279MB
16:45:24: [FileMgr] > System Volume Information
16:45:24: [FileMgr]   WPSettings.dat - 12
16:45:24: [FileMgr] Testseq1.fseq - 720072
16:45:24: [FileMgr] Configuration File: '/config.json' loaded.
16:45:24: [FileMgr] Configuration File: '/input_config.json' loaded.
16:45:24: [FileMgr] Configuration File: '/output_config.json' loaded.
16:45:25: [FileMgr] ERROR: Cannot find 'wificonfig.json' for reading. File does not exist.
16:45:25: [FileMgr] SD file: 'wificonfig.json' not found.
16:45:25: [FileMgr] Configuration File: '/admininfo.json' saved 186 bytes.
16:45:25: [FileMgr] Configuration File: '/config.json' saved 406 bytes.
16:45:25: [WiFiDrv] WiFi Entering State: Connecting Using Config Credentials
16:45:26: [WiFiDrv] Using Static IP
16:45:26: [WiFiDrv] Connecting to 'XMASNET' as DIG-QUAD-TEST1
16:45:26: [WiFiDrv] WiFi Entering State: Connected To AP
16:45:27: [WiFiDrv] Connected with IP: 192.168.2.110
16:45:27: [WebMgr] Web server listening on port 80
16:45:27: [  FPPD] FPPDiscovery subscribed to broadcast messages on port: 32320
16:45:27: [  FPPD] FPPDiscovery subscribed to multicast: 239.70.80.80:32320

DESCRIPTION OF ENVIRONMENT
QuinLED-DIG-QUAD-AE+ running as remote.
 - 4 Ports WS2811 pixels on all ports
Raspberry Pi 4B 8G with FPP software v 7.2 running as player with multisync via broadcast.
 - Test sequence of 3 effects (10 seconds each) with repeat selected.

Random reset of ESP32 during sequence play. Reset happens most frequently at the end of the sequence before repeating.

Please let know if you need additional information or if I need to change the format of the information provided.

Thank you.

MartinMueller2003, 2023-10-11:
Can you attach a sequence that fails?
Also attach a screen shot of the Admin page.

DaleTiger, 2023-10-11:
[image]
[Testseq1.zip](https://github.com/forkineye/ESPixelStick/files/12874306/Testseq1.zip)

onewithhammer, 2023-10-23:
I'm seeing the same issue.  Eventually the unit will stop responding.  I sent you a link to a video on Discord demo'ing the issue.

MartinMueller2003, 2023-10-24:
First issue is the file format is not correct:
[InputMgr] ParseFseqFile:: Could not start. Testseq1.fseq is not a v2 uncompressed sequence

DaleTiger, 2023-10-27:
Sorry about that Martin. Hopefully this one works. Thanks.
[Testseq1_10_27.zip](https://github.com/forkineye/ESPixelStick/files/13193512/Testseq1_10_27.zip)

MartinMueller2003, 2023-10-29:
Please try the dist found here: https://drive.google.com/drive/folders/1bGrkXSMoGjWtX5iWspPFRogM2G6c1VXu?usp=sharing

MartinMueller2003, 2023-10-31:
PR: Remote play updates #680

closed 2023-10-31 by MartinMueller2003

DaleTiger, 2023-11-13:
Hi Martin. Sorry for the very delayed response.  I've been tied up on non-xmas stuff.

I ran the distro you sent me. That it fixed the problem. I looped the sequence 150+ times without an error...

Thanks for your help.