forkineye/ESPixelStick #677: Factory reset leaves the board unusable

marcobrianza, 2023-10-21:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
ESPixelStick v4.0-ci6397314717 (Oct  3 2023 - 19:05:55)

**Hardware Version**
QUINLED DIG OCTA

**Binary release or compiled yourself?**
CI release

**Operating System (and version)**
MAC os 10.15.17

**Web Browser (and version)**
Chrome

**Access Point**
Ubiquity AirRouter

**Describe the bug**
After pressing the button to factory default the board goes in an unusable state.
see attached serial log file

The only way I found to fix this is to reflash from serial

[fork-reset.txt](https://github.com/forkineye/ESPixelStick/files/13060576/fork-reset.txt)

MartinMueller2003, 2023-10-21:
That is expected. When you reset to factory defaults, you destroy the WiFi credentials. The device will go into AP mode allowing you to connect via your phone or tablet. You can then enter new WiFi credentials.

marcobrianza, 2023-10-21:
Im my case the AP mode came up briefly but it quickly went away because the board rebooted as you can see in the logs. 
After further investigation if was a power issue. Powering the dig octa via a USB isolator made that strange deadlock situation :-(
Powering directly does not produce the issue.

closed 2023-10-21 by marcobrianza