forkineye/ESPixelStick #679: FPP Multicast will not work

TheHappyGrizzly, 2023-10-30:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version** v4.0 Beta 3

**Hardware Version** ESP32

**Binary release or compiled yourself?** Java flash tool

**Operating System (and version)** Windows 11

**Web Browser (and version)** Chrome 118.0.5993.118

**Access Point** 

**Describe the bug** I'm getting the FPP sync packets and everything from xSchedule sent to my ESPixelStick in FPP remote mode. I can also play the file directly from my ESPixelStick without any issues. However, for some reason, multicast does not send my controller the sync packets. Broadcast mode will, but xSchedule must not be finding my IP address even when manually added as a FPP remote. The FPP Connect page in xLights does recognize my ESP32 as a FPP remote.

MartinMueller2003, 2023-10-31:
Just ran the latest version against FPP 5 & 7. Multicast works on both

MartinMueller2003, 2023-11-13:
Have you retested using the latest version? If it works I would like to close this issue.

TheHappyGrizzly, 2023-11-15:
I've tried multiple versions and have decided that the issue likely lies in xSchedule itself. Thank you for your help.

closed 2023-11-15 by MartinMueller2003