forkineye/ESPixelStick #685: blank time can't be changed

mxklb, 2023-11-05:
**ESPixelStick Firmware Version**
latest CI build: ESPixelStick_Firmware-4.0-ci6755148587 

**Hardware Version**
espixelstick v3

**Binary release or compiled yourself?**
CI build artifact

**Operating System (and version)**
Linux 5.10.0-26-amd64 SMP Debian 5.10.197-1 (2023-09-29) x86_64 GNU/Linux

**Web Browser (and version)**
Vivaldi 6.4.3160.34 stable

**Access Point**
no

**Describe the bug**

The _Blank Time (s)_ can't be changed. It always defaults to 5.

It's not possible to disable (0) or set it to anything else. This makes ESPixelStick kind of useless as a DMX Proxy, Input E1.31 ->  Output DMX via MAX485. When not sending E1.31 constantly, DMX signal gets always blanked out after 5s :disappointed:  

Expected behavior:

1. _Blank Time (s) = 0_ disable blackout = constant/latest DMX signal output
2. _Blank Time (s) != 5_ blackout happens after the configured time

Workaround: Send E1.31 continuously (= unnecessary/redundant E1.31 WiFI traffic)   

See Screenshot for details:

[image]

Everything else on this page can be saved, only _Blank Time (s)_ can't be changed.

MartinMueller2003, 2023-11-05:
Fixed device config values cant be changed #686

MartinMueller2003, 2023-11-06:
My latest build is always found here: https://github.com/MartinMueller2003/ESPixelStick/releases/tag/dist

mxklb, 2023-11-06:
Tested it: Sadly it's still not working as expected. _Blank Time (s)_ always is 5s nevertheless what value is entered :disappointed: 

@MartinMueller2003 can you please confirm this or how did you test your "fix" to be working as it should be?

closed 2023-11-06 by mxklb