forkineye/ESPixelStick #694: Unable to read fseq file althought it is there

cybercop23, 2023-11-30:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-ci6771207306

**Hardware Version**
DigQuad w/ AE+ hat (8 outputs)
ESP32 - esp32_quinled_quad_ae_plus_8

**Binary release or compiled yourself?**
Released/github

**Operating System (and version)**
Mac Ventura 13.6

**Web Browser (and version)**
Chrome Version 119.0.6045.159 (Official Build) (arm64)

**Access Point**
No

**Describe the bug**
Uploaded sequence using FPP connect and I can see the FSEQ files in File Manager. If I reboot, it will play the current sequence using FPP Remote, but once it moves to the next sequence it gets stuck on 0/0 and shows a message that that it can't read the file.

[image]
[image]

MartinMueller2003, 2023-11-30:
You can put the sd card into a USB SD Card reader and get the file using your PC drive explorer. 

FYI: That image is 2 revs behind but none of the changes would impact this functionality.

MartinMueller2003, 2023-11-30:
No the GUI does not support downloading the file to your PC.

cybercop23, 2023-11-30:
I created a simple FSEQ with only models that are on that ESPixelStickV4. Manually uploaded it and is attached here.
I added that to my playlist in the 2nd slot and kicked off the playlist. First SEQ played fine... as soon as it moved to the new/2nd seqeunce I got the error.

[image]
[TestESPV4.fseq.zip](https://github.com/forkineye/ESPixelStick/files/13513687/TestESPV4.fseq.zip)

cybercop23, 2023-11-30:
> You can put the sd card into a USB SD Card reader and get the file using your PC drive explorer.
> 
> FYI: That image is 2 revs behind but none of the changes would impact this functionality.

Yeah. I looked the commits hoping there was something there but didn't see anything that would have affected this behaviour.

MartinMueller2003, 2023-11-30:
There is a python script that dumps this information 
...\ESPixelStick\tools\fseqinfo.py 
Pass in the fseq path and it will tell you about the file.

cybercop23, 2023-11-30:
Pulled the card and ran the script on all of them (+ they are attached). 
When I ran my script on them, I ran into an issue with spaces in the file name... Can that be the issue? It gets stuck on reading an fseq with space in the filename and then dies....
For my script I manually replaced the spaces with _ 

A_Christmas_Storm.fseq:
  V2.0 sparse uncompressed 3960 channels with 8777 frames @ 25ms from xLights Macintosh 2023.20

Animation7.fseq:
  V2.0 sparse uncompressed 3960 channels with 1200 frames @ 50ms from xLights Macintosh 2023.20

Christmas_Every_Day.fseq:
  V2.0 sparse uncompressed 3960 channels with 4587 frames @ 50ms from xLights Macintosh 2023.20

Just_Yellow.fseq:
  V2.0 sparse uncompressed 3960 channels with 600 frames @ 50ms from xLights Macintosh 2023.20

Reminder.fseq:
  V2.0 sparse uncompressed 3960 channels with 793 frames @ 50ms from xLights Macintosh 2023.21

TestESPV4_manual.fseq:
  V2.2 uncompressed 162848 channels with 600 frames @ 50ms from xLights Macintosh 2023.21

TheEnd.fseq:
  V2.0 sparse uncompressed 3960 channels with 600 frames @ 50ms from xLights Macintosh 2023.21

You_Make_It_Feel_Like_Christmas.fseq:
  V2.0 sparse uncompressed 3960 channels with 6270 frames @ 25ms from xLights Macintosh 2023.20
[Archive.zip](https://github.com/forkineye/ESPixelStick/files/13516014/Archive.zip)

MartinMueller2003, 2023-11-30:
And notice that the ESP files are all "sparse" files. The ESP con support a max of 9000 channels. Your test file can never play correctly on V4.

cybercop23, 2023-11-30:
Thank you very much Marin for your help. Yeah... with only 6 and 7 being supported and do a lot of zoom room support, i keep on the latest. Wondering if the fseq as they are work on your 5.1 setup. there's nothing in the fseq file that will make them not being backwards compatible. Thanks again for your assistance.

MartinMueller2003, 2023-11-30:
I had already verified last summer that there is nothing significantly different in the fseq header that would cause the ESP to have an issue. I did find that the latest version of the sync message did cause us a problem (since fixed) and that the ESP V4 is not sending all of the desired information in the status response messages. There does seem to be an oddity between multisync and broadcast sync messages and on my home system only the broadcast would work properly. FPP 5.1 did not have this issue so I am going to have to dig a little in the spring to see what is going on.

MartinMueller2003, 2023-11-30:
Are you saying that after poking the ESP multiple times and redoing the downloads until they are all correct, that you can play all of the sequences from the FPP. And if you add in that test file, the ESP stops (as it should) but then fails to start on the next correct file?

cybercop23, 2023-11-30:
Ahhh.. did not think of that, but I just tested it that way. Been running for hours w/o any issues reading the files and no errors. I added the TestESPV4 compressed file in rotation and got the compressed message but after that.... I also get the Could not open file... although it has been running for hours... When I remove the bad Test file and restart ESP all is back to normal.

[image]

MartinMueller2003, 2023-12-08:
Let's hope it addresses your issue. :)

cybercop23, 2023-12-11:
Okay... still testing it, but this seems to work so much better. I can even FPP Connect and send fseq files while the show is playing.
There seems to be an issue with xLights not wanting to FPP Connect to other FPPs and ESPixel at the same time, but that's on the xLights side I think.

closed 2023-12-11 by cybercop23