forkineye/ESPixelStick #695: Access throw http by name and not only by IP (eg.  http://ESPixelStick.local )

fedekrum, 2023-12-01:
**ESPixelStick Firmware Version**
[   ESP] ESPixelStick v4.0-ci6981356867 (Nov 24 2023 - 13:31:42)
[   ESP] SDK:2.2.2-dev(38a443e)/Core:3.0.2=30002000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:6105635

**Hardware Version**
Node MCU v3

**Binary release or compiled yourself?**
Binary

**Operating System (and version)**
Macos Catalina 10.15.7 (19H2026)

**Web Browser (and version)**
chrome latest

**Access Point**

**Describe the bug**
Not sure if it is a bug or a feature request.
It would be great to access ESPixelStick not by IP address only but by its network QDN ( http://espixelstick.local or similar)

MartinMueller2003, 2023-12-01:
This is already available. The hostname field in the network settings defines the network name used to resolve the PixelSticks IP address.

fedekrum, 2023-12-02:
Thanks @MartinMueller2003 . It is not working for me now, but I will try connecting with other router and report here.