forkineye/ESPixelStick #696: No output after moving from 4.0 beta 4 to 4.0 beta 5

KHouse75, 2023-12-02:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
ESPixelStick-4.0-beta5

**Hardware Version**
Custom hardware designed for a 4MB flash esp8266 NodeMCU Development board

**Binary release or compiled yourself?**
Binary release

**Operating System (and version)**
Windows 11

**Web Browser (and version)**
Chrome 119.0.6045.200

**Access Point**
TP-Link Omada AC1350

**Describe the bug**
I've been running previous versions of ESPixelStick with no issues.  4.0 Beta 4 and previous releases work out of the box.  I saw that Beta 5 was just released so I installed it on my esp8266 NodeMCU.

I'm using WS2811 strands.  Upon power on, The first 12 lights or so light solid white and remain lit.  I can connect to the web interface with no issues.  When I enable the effects option, the LEDs do not light.  When I try to use mqtt to play a sequence file from the sd card, the LEDs do not do anything.  When I send E1.31 data, nothing happens with the lights but I can see that it's receiving the data through the web interface.  When I turn on an effect through home assistant, nothing happens.  The first 12 light remain lit the entire time no matter when I do.

I also noticed that the diagnostics page does not display the pixel colors even when the secondary input effects option is selected.

I programmed another NodeMCU that's from a different manufacturer and it does the exact same thing.

I flashed 4.0 Beta 4 back onto the board and everything works as expected.

DShaeffer, 2023-12-02:
I am seeing this same thing. I have an official ESPixelStick V3 which I just purchased. The only difference is the first light is off, the next 11 are on, and the remainder are off. I can connect in xlights and play the sequence using DPP. However, I can no longer get the player to run the sequence when I set it to play an individual file.

KHouse75, 2023-12-02:
As another test, I flashed an esp32 with the 4.0 beta 5 and the same issue occurs.  When I flash the same esp32 with 4.0 beta 4, everything works as expected.

KHouse75, 2023-12-02:
> I am seeing this same thing. I have an official ESPixelStick V3 which I just purchased with the esp8266. The only difference is the first light is off, the next 11 are on, and the remainder are off. I can connect in xlights and play the sequence using DPP. However, I can no longer get the player to run the sequence when I set it to play an individual file. The espixelstick interface indicates that it is playing the file but nothing on the pixels. Also, no data is shown on the diagnostics screen.

Thanks for confirming the issue.  It may by 11 with the first one off on mine as well but it was was dark out and pouring down rain and I was about 30 ft away trying to count from my window :) .

KHouse75, 2023-12-02:
> I am new to this so still figuring this out. However, it seems to be related to the sequence files. I created a new sequence file with one 10 second effect and that worked fine on the espixelstick player and via FFP on a Raspberry Pi. The other more complex sequence will not play or will light up 10-12 pixels.

If you enable effects as your secondary input, do the pixels light up properly?

DShaeffer, 2023-12-02:
I did eventually get my sequence to play by opening a new instances of xLights, copying and pasting the effects from the non-working xLights project/sequence to the new project, and then reuploading via FPP connect. This was my first sequence and I couldn't tell you why the sequence became corrupted or how to fix it. You might try a new simple sequence to debug your issue.

KHouse75, 2023-12-03:
I was able to get this working but I don't really know where the problem resides.

Before I got it working, I decided to hard wire directly to a short strip of 5V WS2812B pixel strip.  23 pixels total.  The same problem occurred.  First pixel off, next 12 bright white.  For some reason, I decided to try a different output (GECE) and when I did, the 12 lights turned off and the pixels turned the correct color.  It looks like, when WS2811 is selected, it's actually disabled that's selected.

I then decided to do a complete flash erase using esptool then reflashed the esp8266 with 4.0 Beta 5.  I then configured all the settings and everything worked as expected.  What's strange is I previously flashed beta 5 on brand new, freshly opened ESP32 and ESP8266 devices and they both has the same issue.  I even opened them in a different browser and opened them in a private windows to eliminate any caching.

Next, I'll be seeing if the same thing fixes the esp32.