forkineye/ESPixelStick #697: Reproducible packet error

fedekrum, 2023-12-02:
**ESPixelStick Firmware Version**
FW Version 4.0-ci7063024905
Build Date Dec 1 2023 - 17:15:45

**Hardware Version**
Architecture ESP8266
Board Name espsv3 (Node MCU 1.0 v3)

**Binary release or compiled yourself?**
Binary release

**Operating System (and version)**
Macos Catalina

**Web Browser (and version)**
Chrome

**Access Point**
FritzBox 7490

**Describe the bug**
I can reproduce a Packet Error, but I am not sure if ESPixelStick is responsible for it or the sender.
I have made a script in py using lib PyArtNet. 
https://github.com/fedekrum/DMX/blob/main/scanDMX.py

usage: scanDMX.py [-h] -a ADDRESS -u UNIVERSE [-c CHANNEL] [-r RANGE]
                  [-s SPEED] [-k]

Scans DMX channels with PyArtNet.

options:
  -h, --help            show this help message and exit
  -a ADDRESS, --address ADDRESS
                        IP address or hostname of the sACN node
  -u UNIVERSE, --universe UNIVERSE
                        DMX universe number (1-63999)
  -c CHANNEL, --channel CHANNEL
                        DMX channel number (1-512), optional
  -r RANGE, --range RANGE
                        Range of channels to animate (1-512), optional
  -s SPEED, --speed SPEED
                        Animation speed in milliseconds (default 500ms)
  -k, --key-pause       Require key press to start animation for each channel

**How to reproduce**
1) Start fresh your ESPixelStick and notice **E1.31 Status** - **Packet Error** and **Total Packets** in zero.

[image]

2) Execute the script one time only
python3 scanDMX.py -a192.168.188.81 -u1 -c1
You will notice that xxx Total Packets were sent and no Packer Error will occur.

[image]

3) Execute the script once more
Notice that this time you get a Packet Error

[image]

4) If you keep on repeating the execution you get one packet error each time, even if you change the command parameters to do an extensive scan.

[image]

Thanks in advance

fedekrum, 2023-12-04:
I have simplified what the program does to have fewer packets to analyze (6 packets only).
I added Wireshark packets as a text report, .pcap and .pcapng.

fedekrum, 2023-12-05:
Thanks @MartinMueller2003 
So I assume that inside the E1.31 data, there is a sequential step number in each packet, right?
And the problem is that my program, each time it gets executed starts from step 0 (or 1) each time so the ESPixelStick gets that as a packet error, right?
Can you confirm if my interpretation is good?

If so, it should be more of a warning than a packet error.
I understand a packet error more as a packet loss because it is malformed.

MartinMueller2003, 2023-12-05:
Yes you are correct and yes it is a packet error. We cant tell if something dropped or got reset or was restarted or if a packet got delayed and then suddenly arrived late. So the simple to implement rule is: "Not the expected number? Increment error count".

MartinMueller2003, 2023-12-06:
can I close this as "not a bug"?

fedekrum, 2023-12-06:
Absolutely and thanks!!

closed 2023-12-07 by MartinMueller2003