forkineye/ESPixelStick #698: QUINLET DIG-OCTA - several output types missing in device setup

DaleTiger, 2023-12-03:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-beta5

**Hardware Version**
QuinLED Dig-Octa

**Binary release or compiled yourself?**
Binary release

**Operating System (and version)**
Windows 11

**Web Browser (and version)**
Mictosoft Edge  Version 119.0.2151.97

**Access Point**
Netgear AC1450

**Describe the bug**
Several output types missing in device setup for DIG-OCTA firmware
GECE, DMX, Renard, Serial  (I need GECE for my setup).

MartinMueller2003, 2023-12-05:
What output types do you believe are missing? The current supported list was chosen by Quindor. If you really think something is missing, then do a factory reset. Thaat usually clears such things up.

MartinMueller2003, 2023-12-06:
And if you want GECE (Quindor asked not to include that) I can give you a build that includes it.

DaleTiger, 2023-12-07:
Thanks Martin.

It's strange they didn't want to include GECE on the OCTA since it's supported on the DIG-QUAD.

Sincerely,

Dale Smith

DaleTiger, 2023-12-08:
> Please download the latest dist in my google drive:
> https://drive.google.com/drive/folders/1bGrkXSMoGjWtX5iWspPFRogM2G6c1VXu?usp=sharing

Martin,

  I really appreciate this!  I still have a lot of GECE wire frame props and pathway lights.

Is there a process to contact the folks at Quin-LED to talk about their pixel type support criteria?  I would really like to get their thoughts on pixel type support.
There are a couple of other pixel types I don’t use at this time but could see a future need for them (i.e. Renard, Serial) to name a few.

I would also like to offer any assistance you may need on program development (like beta testing) or whatever would help further the cause.

Again, thanks for the great work you do.

Sincerely,

Dale Smith

MartinMueller2003, 2023-12-08:
I got my direction from Quindor. You can contact him on his discord server: Intermit.tech. I also do not think a lot of thought went into the decision. The focus for the octo was on specific pixel types so all serial protocols were left out.

DaleTiger, 2023-12-16:
Good afternoon Martin,

  I hope you don’t mind that I use e-mail to ask this question. I don’t think it’s a bug so I thought e-mail would be best.

Running 4.0-beta5

My question is this. What does this message from the serial output of a QuinLED-DIG-QUAD mean?

13:04:49: [InputMgr] ParseFseqFile:: Could not start: XmasEve12-24.fseq File does not contain enough data to meet the Stated Channel Count * Number of Frames value. Expected 11381832, Got: 2420736

Thanks for your help.

Sincerely,

Dale Smith

MartinMueller2003, 2023-12-16:
I dont mind at all. 

It means the fseq file has been truncated in some way. The fseq header at the start of the file says there should be N bytes of data in the file but the file actually has less than that. This is an oft reported issue when using xLights FPP Connect to send files to the ESP.
Martin

MartinMueller2003, 2023-12-20:
Output Types have been added.

closed 2023-12-20 by MartinMueller2003