forkineye/ESPixelStick #699: 4.0b5 empty Home screen

Cassock, 2023-12-05:
**ESPixelStick Firmware Version**
4.0 beta 5

**Hardware Version**
Espixelstick V3 from Shelby

**Binary release or compiled yourself?**
Binary release from 3 days ago

**Operating System (and version)**
Win11

**Web Browser (and version)**
Chrome 119 (same in Edge)

**Access Point**
?

**Describe the bug**
Flashed 2 units earlier today and both have an empty Home page as depicted below. Looks like all other functionality is working fine, so just a UI issue along with a bunch of errors in the console. 
[image]
[image]

fedekrum, 2023-12-05:
What I do is to nuke all the ESP8266 data first by uploading an empty script through Arduino IDE.
(There should be more clever ways to do this)
I had problems before reloading ESPixelStick.

MartinMueller2003, 2023-12-05:
Do the other pages work? Are there error messages coming out on the ESP console?

Cassock, 2023-12-06:
All other pages work fine. I just reflashed and it appears the Home stats show up fine until I set the secondary input to FPP remote, then get a number of [WebMgr] ERROR: Status Doc is too small. How would you like me to send you logs? I can't attach them here

Cassock, 2023-12-06:
[120623 logs.txt](https://github.com/forkineye/ESPixelStick/files/13591092/120623.logs.txt)

MartinMueller2003, 2023-12-06:
This is the critical error:

[WebMgr] ERROR: Status Doc is too small

MartinMueller2003, 2023-12-08:
Just reproduced this issue.

MartinMueller2003, 2023-12-08:
Fixed in my latest PR. google drive dist has been updated.

closed 2023-12-08 by MartinMueller2003