forkineye/ESPixelStick #701: Dig Quad and others temperature sensor

stijn-dhondt, 2023-12-06:
**Is your feature request related to a problem? Please describe.**
Not related to a problem

**Describe the solution you'd like**
The newer series of dig quad's and even uno's can have a temperature sensor on them.
Would be nice to have the readout from these

**Additional context**
Dont think the octa's have these but maybe an external sensor so we can track the temps in the boxes would be nice...

MartinMueller2003, 2023-12-06:
Would you like to add it?

stijn-dhondt, 2023-12-21:
Hi @MartinMueller2003,

Would love to but my programming skills are like bellow zero Don't know if anyone else could help... Not that urgent but havng an eye on temps is nice to have if the sensor is already there...

Sorry!

MartinMueller2003, 2024-01-02:
Almost done for the DIG Quad. Are any of the other DIG products using the same sensor?

MartinMueller2003, 2024-01-02:
Added Temperature Sensor for DigQuad PR #723

closed 2024-01-02 by MartinMueller2003

stijn-dhondt, 2024-01-12:
@MartinMueller2003 if i'm right than you assigned IO pin 13 to this but accoring to pinout bellow it should be IO18... 
https://quinled.info/2020/07/02/quinled-dig-quad-pinout/
Also the Dig uno has the same sensor.

Thx for the work already!

Stijn

MartinMueller2003, 2024-01-13:
Updated the default GPIO Values 
Added Dig-Uno