forkineye/ESPixelStick #702: [FileMgr] ***ERROR: JSON Doc too small to hold the list of SD files***

Terrevue, 2023-12-06:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
ESPixelStick v4.0-beta5 (Dec  1 2023 - 17:11:40)

**Hardware Version**
QuinLED Quad AE+

**Binary release or compiled yourself?**
Binary release

**Operating System (and version)**
Used both Mac 11.6.4 and Windows 11 Pro

**Web Browser (and version)**
n/a

**Access Point**
n/a

**Describe the bug**
First time the web interface is accessed with an SD card plugged in, the following error is received:

12:08:52: [FileMgr] ***ERROR: JSON Doc too small to hold the list of SD files***
12:09:03: [FileMgr] ***ERROR: JSON Doc too small to hold the list of SD files***
12:09:06: [FileMgr] ***ERROR: JSON Doc too small to hold the list of SD files***
12:09:08: E (33348) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
12:09:11: E (33348) task_wdt:  - async_tcp (CPU 0/1)
12:09:11: E (33348) task_wdt: Tasks currently running:
12:09:11: E (33348) task_wdt: CPU 0: IDLE
12:09:11: E (33348) task_wdt: CPU 1: loopTask
12:09:11: E (33348) task_wdt: Aborting.
12:09:11: 
12:09:11: abort() was called at PC 0x40115034 on core 0
12:09:11: 
12:09:11: 
12:09:11: Backtrace:0x40084181:0x3ffbeaac |<-CORRUPTED
12:09:11: 
12:09:11: 
12:09:11: 
12:09:11: 
12:09:11: ELF file SHA256: 0000000000000000
12:09:11: 
12:09:11: Rebooting...
12:09:11: ets Jul 29 2019 12:21:46
12:09:11: 
12:09:11: rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
12:09:11: configsip: 0, SPIWP:0xee
12:09:11: clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
12:09:11: mode:DIO, clock div:2
12:09:11: load:0x3fff0030,len:1184
12:09:11: load:0x40078000,len:12784
12:09:11: load:0x40080400,len:3032
12:09:11: entry 0x400805e4

I have tried other OS's, different SD cards, reformatting, creating volumes of only 32gb.  No matter what, when I add an SD card, this occurs.  If I remove the SD card, everything boots fine.

MartinMueller2003, 2023-12-06:
How many files are on the SD card?

Terrevue, 2023-12-06:
As a follow up, the same error occurs with a custom compiled version for a seperate ESP32 hardware, based on a source version from Nov.

12:37:04: [   ESP] ESPixelStick v4.x-dev (Nov 16 2023 - 12:45:01)
12:37:04: [   ESP] v4.4.1-1-gb8050b365e

I had issues last year as well that was only resolved by pulling the SD card.  I assumed that I had cheap cards so didn't investigate further.  This year I'm using high endurance cards and the issue is still occuring, it appears.

Terrevue, 2023-12-06:
34 files total

Terrevue, 2023-12-06:
Here is the list of files, in case there is a naming issue involved:

pentatonix - Little Drummer Boy - 20231204.fseq
linus and lucy-20231204.fseq
chipmunk remix - 20231204.fseq
Whats This - Normalized - 20231204.fseq
Underneath the Tree - 20231204.fseq
U2 - Christmas (Baby Please Come Home) - 20231204.fseq
Turn Up Ye Merry Gentlemen - 20231204.fseq
Snow Miser and Heat Miser - 20231204.fseq
Santa Claus is Coming To Town - Sugar Plumb Fairy mix-20231204.fseq
Ramones - Merry Christmas (I Don't Want to Fight Tonight) (Official Music Video).fseq
Polar Express Mashup - 20231204.fseq
Pennies From Heaven-20231204.fseq
Nothing For Christmas - 20231204.fseq
My Favorite Things - Pentatonix - 20231204.fseq
Mad Russian-20231204.fseq
Little Drummer Boy (Live) - 20231204.fseq
Let It Go-20231204.fseq
John Legend - What Christmas Means to Me (Official Audio) ft. Stevie Wonder - with faces - 20231204.fseq
John Legend - What Christmas Means to Me (Official Audio) ft. Stevie Wonder - 20231204.fseq
It Must've Been Ol Santa Claus - 20231204.fseq
Idina Menzel Michael Buble - Baby Its Cold Outside - 20231204.fseq
Holiday Road (National Lampoon's Vacation) - 20231204.fseq
Have Yourself A Merry Little Christmas.fseq
Have Yourself A Merry Little Christmas 2.fseq
Happy Christmas.fseq
Gwen Stefani - You Make It Feel Like Christmas ft. Blake Shelton - 20231204.fseq
Eagles -Please Come Home for Christmas - 20231204.fseq
Do they Know it's Christmas ~ Band Aid 1984 - 20231204.fseq
Christmas Medley 2018.fseq
Cantique De Noel (O Holy Night) - 20231204.fseq
AfterHours.fseq
7 song medley - 20231204.fseq
2019 Ariana Grande - Santa Tell Me - 20231204.fseq
08 - Carol of the Bells - 20231204.fseq

MartinMueller2003, 2023-12-06:
Dont think I put that many files on a card. Most shows have 5-8 sequences. Looking at the code, if the total number of file name characters exceeds 2000 bytes then you will get this error.

Terrevue, 2023-12-06:
I've always done an individual sequence per song.  Is there any way to expand that limit?

Terrevue, 2023-12-06:
Total string len of all characters is 1,582

MartinMueller2003, 2023-12-06:
I am looking at the list. It looks like there are some ling file names (105B) and parenthesis. Can you zip the drive and upload the image to this dir:
https://drive.google.com/drive/folders/1bGrkXSMoGjWtX5iWspPFRogM2G6c1VXu?usp=drive_link

Terrevue, 2023-12-06:
Just tried and it doesn't appear I have permissions to upload.

MartinMueller2003, 2023-12-06:
Done.

I am in the process of rebuilding dist.zip with 25% more filename space. This uses a lot of ram internally so I try to avoid grabbing a large amount in one chunk. The real fix may be to chunk the filename list upload.

Terrevue, 2023-12-06:
I just remembered that I can't use XLight's FPP upload because it always crashes.  The crash dump just locks up so I just kill the process and manually copy the files to each card (way faster anyway).  I never thought anything of it because I always just expect the occasional bug in software.  I'll bet these events are related.  Now, to figure out which file name it was...

Terrevue, 2023-12-06:
The JSON error is gone but it's still rebooting.  I removed anything with an apostrophe to be safe and either that or the number of files caused the JSON error to go away.  Here is what it states now (Including the reboot file load so you'll see the files currently being loaded)

13:26:55: E (73979) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
13:27:09: E (73979) task_wdt:  - async_tcp (CPU 1)
13:27:09: E (73979) task_wdt: Tasks currently running:
13:27:09: E (73979) task_wdt: CPU 0: IDLE
13:27:09: E (73979) task_wdt: CPU 1: loopTask
13:27:09: E (73979) task_wdt: Aborting.
13:27:09: 
13:27:09: abort() was called at PC 0x40115e84 on core 0
13:27:09: 
13:27:09: 
13:27:09: Backtrace:0x400840b1:0x3ffbea9c |<-CORRUPTED
13:27:09: 
13:27:09: 
13:27:09: 
13:27:09: 
13:27:09: ELF file SHA256: 0000000000000000
13:27:09: 
13:27:09: Rebooting...
13:27:09: ets Jun  8 2016 00:22:57
13:27:09: 
13:27:09: rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:27:09: configsip: 0, SPIWP:0xee
13:27:09: clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:27:09: mode:DIO, clock div:2
13:27:09: load:0x3fff0030,len:1184
13:27:09: load:0x40078000,len:12784
13:27:09: load:0x40080400,len:3032
13:27:09: entry 0x400805e4
13:27:09: 
13:27:09: [   ESP] ESPixelStick v4.x-dev (Nov 16 2023 - 12:45:01)
13:27:09: [   ESP] v4.4.1-1-gb8050b365e
13:27:09: [FileMgr] Flash file system initialized. Used = 233472 out of 327680
13:27:09: [FileMgr] Listing directory: /
13:27:09: [FileMgr] 'UpdRecipe.json': 	'784'
13:27:09: [FileMgr] 'admininfo.json': 	'165'
13:27:09: [FileMgr] 'config.json': 	'558'
13:27:09: [FileMgr] 'input_config.json': 	'1994'
13:27:09: [FileMgr] 'output_config.json': 	'20519'
13:27:09: [FileMgr] Listing directory: //www
13:27:10: [FileMgr] 'alexa.html.gz': 	'146'
13:27:10: [FileMgr] 'apa102.html.gz': 	'1503'
13:27:10: [FileMgr] 'artnet.html.gz': 	'385'
13:27:10: [FileMgr] 'ddp.html.gz': 	'144'
13:27:10: [FileMgr] 'dmx.html.gz': 	'455'
13:27:10: [FileMgr] 'e1_31.html.gz': 	'460'
13:27:10: [FileMgr] 'effects.html.gz': 	'973'
13:27:10: [FileMgr] 'esps.css.gz': 	'13424'
13:27:10: [FileMgr] 'esps.js.gz': 	'61179'
13:27:10: [FileMgr] 'favicon.ico': 	'6450'
13:27:10: [FileMgr] 'fpp_remote.html.gz': 	'348'
13:27:10: [FileMgr] 'gece.html.gz': 	'501'
13:27:10: [FileMgr] 'gs8208.html.gz': 	'1579'
13:27:10: [FileMgr] 'index.html.gz': 	'4948'
13:27:10: [FileMgr] 'mqtt.html.gz': 	'656'
13:27:10: [FileMgr] 'relay.html.gz': 	'507'
13:27:10: [FileMgr] 'renard.html.gz': 	'528'
13:27:10: [FileMgr] 'serial.html.gz': 	'619'
13:27:10: [FileMgr] 'servo_pca9685.html.gz': 	'523'
13:27:10: [FileMgr] 'tls3001.html.gz': 	'1503'
13:27:10: [FileMgr] 'tm1814.html.gz': 	'1617'
13:27:10: [FileMgr] 'ucs1903.html.gz': 	'1580'
13:27:10: [FileMgr] 'ucs8903.html.gz': 	'1585'
13:27:10: [FileMgr] 'ws2801.html.gz': 	'1502'
13:27:10: [FileMgr] 'ws2811.html.gz': 	'1582'
13:27:10: [FileMgr] SD Card Size: 61120MB
13:27:10: [FileMgr] > System Volume Information
13:27:10: [FileMgr]   WPSettings.dat - 12
13:27:10: [FileMgr] My Favorite Things - Pentatonix - 20231204.fseq - 10420369
13:27:10: [FileMgr] Nothing For Christmas - 20231204.fseq - 7573398
13:27:10: [FileMgr] Pennies From Heaven-20231204.fseq - 11224543
13:27:11: [FileMgr] pentatonix - Little Drummer Boy - 20231204.fseq - 35053636
13:27:11: [FileMgr] Polar Express Mashup - 20231204.fseq - 17051933
13:27:11: [FileMgr] Santa Claus is Coming To Town - Sugar Plumb Fairy mix-20231204.fseq - 20587857
13:27:11: [FileMgr] Snow Miser and Heat Miser - 20231204.fseq - 10868484
13:27:11: [FileMgr] Turn Up Ye Merry Gentlemen - 20231204.fseq - 6906754
13:27:11: [FileMgr] U2 - Christmas (Baby Please Come Home) - 20231204.fseq - 4167549
13:27:11: [FileMgr] Underneath the Tree - 20231204.fseq - 15383091
13:27:11: [FileMgr] Whats This - Normalized - 20231204.fseq - 19916023
13:27:11: [FileMgr] 7 song medley - 20231204.fseq - 62811431
13:27:11: [FileMgr] 08 - Carol of the Bells - 20231204.fseq - 15784630
13:27:11: [FileMgr] 2019 Ariana Grande - Santa Tell Me - 20231204.fseq - 8278884
13:27:11: [FileMgr] AfterHours.fseq - 1053257
13:27:11: [FileMgr] Cantique De Noel (O Holy Night) - 20231204.fseq - 25959893
13:27:11: [FileMgr] chipmunk remix - 20231204.fseq - 6734835
13:27:11: [FileMgr] Eagles -Please Come Home for Christmas - 20231204.fseq - 8107713
13:27:11: [FileMgr] Gwen Stefani - You Make It Feel Like Christmas ft. Blake Shelton - 20231204.fseq - 9809400
13:27:11: [FileMgr] Happy Christmas.fseq - 4241398
13:27:11: [FileMgr] Have Yourself A Merry Little Christmas.fseq - 22009062
13:27:11: [FileMgr] Idina Menzel Michael Buble - Baby Its Cold Outside - 20231204.fseq - 19734624
13:27:12: [FileMgr] It Must've Been Ol Santa Claus - 20231204.fseq - 14206862
13:27:12: [FileMgr] John Legend - What Christmas Means to Me (Official Audio) ft. Stevie Wonder - 20231204.fseq - 14816227
13:27:12: [FileMgr

MartinMueller2003, 2023-12-06:
I suspect it is taking too long to build the list and then transmit it. I am going to have to chunk the response.

Terrevue, 2023-12-06:
It stopped crashing at 15 files:

14:14:55: [FileMgr] My Favorite Things - Pentatonix - 20231204.fseq - 10420369
14:14:55: [FileMgr] Nothing For Christmas - 20231204.fseq - 7573398
14:14:55: [FileMgr] Pennies From Heaven-20231204.fseq - 11224543
14:14:55: [FileMgr] pentatonix - Little Drummer Boy - 20231204.fseq - 35053636
14:14:55: [FileMgr] Polar Express Mashup - 20231204.fseq - 17051933
14:14:55: [FileMgr] Snow Miser and Heat Miser - 20231204.fseq - 10868484
14:14:55: [FileMgr] Turn Up Ye Merry Gentlemen - 20231204.fseq - 6906754
14:14:55: [FileMgr] Underneath the Tree - 20231204.fseq - 15383091
14:14:55: [FileMgr] Whats This - Normalized - 20231204.fseq - 19916023
14:14:55: [FileMgr] 7 song medley - 20231204.fseq - 62811431
14:14:55: [FileMgr] 08 - Carol of the Bells - 20231204.fseq - 15784630
14:14:55: [FileMgr] 2019 Ariana Grande - Santa Tell Me - 20231204.fseq - 8278884
14:14:55: [FileMgr] AfterHours.fseq - 1053257
14:14:55: [FileMgr] chipmunk remix - 20231204.fseq - 6734835
14:14:55: [FileMgr] Eagles -Please Come Home for Christmas - 20231204.fseq - 8107713
14:14:55: [FileMgr] Happy Christmas.fseq - 4241398
14:14:55: [FileMgr] Have Yourself A Merry Little Christmas.fseq - 22009062
14:14:55: [FileMgr] Idina Menzel Michael Buble - Baby Its Cold Outside - 20231204.fseq - 19734624
14:14:56: [FileMgr] Let It Go-20231204.fseq - 11518954
14:14:56: [FileMgr] linus and lucy-20231204.fseq - 4408441

Terrevue, 2023-12-06:
Below is the final list:

18 total real files (files that would be read into memory)
272,270,971 bytes (total that caused crash)
255,206,750 bytes (total stable)
* 21 total files on disk (Mac adds this pointer to deleted files)
** This is the last file I added back before it started crashing again.

Snow Miser and   Heat Miser - 20231204.fseq | 12/5/2023, 8:08:34 AM | 10,868,484
-- | -- | --
* )._Snow Miser and Heat Miser -   20231204.fseq | 12/6/2023, 9:20:20 AM | 4,096
Underneath the Tree - 20231204.fseq | 12/5/2023, 8:10:34 AM | 15,383,091
*) ._Underneath the Tree - 20231204.fseq | 12/6/2023, 9:22:26 AM | 4,096
**) Polar Express Mashup - 20231204.fseq | 12/5/2023, 8:07:02 AM | 17,051,933
*) ._Polar Express Mashup - 20231204.fseq | 12/6/2023, 9:33:48 AM | 4,096
My Favorite Things - Pentatonix -   20231204.fseq | 12/5/2023, 8:04:46 AM | 10,420,369
Nothing For Christmas - 20231204.fseq | 12/5/2023, 8:05:32 AM | 7,573,398
Pennies From Heaven-20231204.fseq | 12/5/2023, 8:06:20 AM | 11,224,543
pentatonix - Little Drummer Boy -   20231204.fseq | 12/5/2023, 8:14:12 AM | 35,053,636
7 song medley - 20231204.fseq | 12/5/2023, 7:33:42 AM | 62,811,431
08 - Carol of the Bells - 20231204.fseq | 12/5/2023, 7:31:14 AM | 15,784,630
2019 Ariana Grande - Santa Tell Me -   20231204.fseq | 12/5/2023, 7:31:54 AM | 8,278,884
AfterHours.fseq | 12/5/2023, 7:34:06 AM | 1,053,257
chipmunk remix - 20231204.fseq | 12/5/2023, 8:11:44 AM | 6,734,835
Eagles -Please Come Home for Christmas -   20231204.fseq | 12/5/2023, 7:40:18 AM | 8,107,713
Happy Christmas.fseq | 12/5/2023, 7:41:38 AM | 4,241,398
Have Yourself A Merry Little   Christmas.fseq | 12/5/2023, 7:43:24 AM | 22,009,062
Idina Menzel Michael Buble - Baby Its   Cold Outside - 20231204.fseq | 12/5/2023, 7:44:28 AM | 19,734,624
Let It Go-20231204.fseq | 12/5/2023, 7:48:52 AM | 11,518,954
linus and lucy-20231204.fseq | 12/5/2023, 8:12:04 AM | 4,408,441

MartinMueller2003, 2023-12-06:
Thanks for the detailed tests. That will help a lot in isolating the issue.

MartinMueller2003, 2023-12-06:
Just uploaded a version of the code that expands the buffer to 4KB.

MartinMueller2003, 2023-12-07:
That is not a fix. It is a work around. I am working on a fix but it will take a bit longer.

Terrevue, 2023-12-08:
Confirmed.  Happens when I access Device of File Manager.  See attached for full log file.  Please let me know if there are any specific tests you'd like me to try.
[QuinLED02.txt](https://github.com/forkineye/ESPixelStick/files/13619963/QuinLED02.txt)

MartinMueller2003, 2023-12-08:
Not sure where it is crashing. I just added a wdt update in the loop just in case we are taking too much time. New dist in same location.

Terrevue, 2023-12-08:
Headed into show time now so I likely won't be able to test until the morning.  I'll send the logs as soon as I have them.

Terrevue, 2023-12-14:
Been a crazy week so I haven't had a chance to test yet.  I hope to have it done by at least tomorrow though.

closed 2023-12-20 by MartinMueller2003