forkineye/ESPixelStick #705: D1 Mini Pro 16MB Flash not working after flashing

fedekrum, 2023-12-11:
**ESPixelStick Firmware Version**
D1 Mini Pro 16MB
**Hardware Version**
D1 Mini Pro 16MB

**Binary release or compiled yourself?**
No

**Operating System (and version)**
MacOS Catalina

**Web Browser (and version)**
Chrome Latest release

**Access Point**

**Describe the bug**
I have bought a D1 Mini Pro 16MB Flash.
After flashing the 16MB version, I can't access the ESPixelStick webpage.
But if I flash the 4MB version, it works.

[Logs using D1 Mini PRO.log](https://github.com/forkineye/ESPixelStick/files/13640361/Logs.using.D1.Mini.PRO.log)
[Logs using D1 Mini PRO 16MB Flash.log](https://github.com/forkineye/ESPixelStick/files/13640362/Logs.using.D1.Mini.PRO.16MB.Flash.log)

MartinMueller2003, 2023-12-11:
It is working as designed. For the 16MB version you did not set an SSID or network passphrase. It defaults to AP mode and after a while it reboots. Try connecting to the device while it is AP mode: 
0:47:52: [WiFiDrv] WiFi Failed to connect using default Credentials
20:48:08: [WiFiDrv] WiFi Entering State: Connecting As AP
20:48:08: [WiFiDrv] WiFi SOFTAP:       ssid: 'ESPixelStick-esps-6428a8
20:48:08: [WiFiDrv] WiFi SOFTAP: IP Address: '(IP unset)

The set an SSID and passphrase. Leave the device connected to the flashtool while you do this so I can see if there are any errors. 
Also, please make sure you are using the dist with a date newer than Dec 9 2023

fedekrum, 2023-12-11:
Hi Martin.
I did the same config for both, only changing the Hardware option.

MartinMueller2003, 2023-12-11:
But the log is showing that the config file has no SSID entry. I have uploaded to my 16MB version using VS Code. Let me try using the flashtool.

fedekrum, 2023-12-11:
Here is a screenshot of my browser after flashing using 192.168.4.1 connecting to the ESPixelStick wifi network.

MartinMueller2003, 2023-12-11:
Thanks. I cant seem to find my D1 Mini Pro (had one but it seems to have grown legs). There are two programming profiles. One for the pro and another for the pro with 16MB. Have you tried both?

fedekrum, 2023-12-11:
yes! With both as explained in the first post.
I wish I could do a march memory test on that 16MB flash to clarify if the problem is a bug or the board.

MartinMueller2003, 2023-12-11:
The 16MB Flash image does not work
The D1 Mini Pro Image DOES work.
[log.txt](https://github.com/forkineye/ESPixelStick/files/13641801/log.txt)

fedekrum, 2023-12-13:
hI Martin.
Is this going to be corrected?
Thanks

MartinMueller2003, 2023-12-13:
I am going to delete the 16mb image since it is the same as the mini pro image. The mini pro image loads and runs on my 16MB mini pro

fedekrum, 2023-12-14:
is there any advantage on the 16MB hardware for this project?

MartinMueller2003, 2023-12-14:
Nope

closed 2023-12-20 by MartinMueller2003