forkineye/ESPixelStick #708: DDP OUTPUT

dosipod, 2023-12-14:
If it would be possible to use DDP OUTPUT  then that would be great . If not possible then a hint on area to change in code  is acceptable

MartinMueller2003, 2023-12-14:
DDP Output? The ESP supports DDP as an input protocol. I am not sure it is useful to have the ESP output DDP. The ESP V4 is not meant to be a show player.

dosipod, 2023-12-14:
I am aware of DDP as an input protocol and works fine , and it could be true that ESP V4 is not meant to be a show player. But thinking about how many Pis could be saved if it is and it has already been done with other open source projects like wled for DDP and ART-NET output . Could be a good uses cases if only price wise for say an esp32 cam as an fpp replacement . I could still hack in code  but not really used to this project  so I just need a little push

MartinMueller2003, 2023-12-14:
you should be able to add it pretty simply. The difficult part is managing which channels are sent to a specific endpoint and adding time awareness to the ESP V4. I have carefully stayed away from using the ESP as a show player because it would be using WiFi in a double transition mode that will add a lot of jitter to the output stream.