forkineye/ESPixelStick #709: Store [Diagnostics] tab settings

fedekrum, 2023-12-15:
**Is your feature request related to a problem? Please describe.**
The problem is that each time I access the [Diagnostics] tab after the page reload, "Display As" (RGB, RGBW, Channels) and "Columns" are reset.

**Describe the solution you'd like**
I imagine 2 solutions.
1) Write in the ESP flash these 2 parameters each time a change is made to one of those.
2) Save them on a browser cookie for a long time each time these are changed

MartinMueller2003, 2023-12-27:
Changes for this are in the latest pull request.

closed 2023-12-27 by MartinMueller2003

fedekrum, 2023-12-27:
Which way you went? Cookies or saving values on ESP flash?(or other...)

MartinMueller2003, 2023-12-27:
cookies. You can test it in my latest dist.zip file.

fedekrum, 2023-12-27:
Can you post your release's URL?

MartinMueller2003, 2023-12-27:
https://github.com/MartinMueller2003/ESPixelStick/releases
Use the dist.zip file you find there. The source code is out of date.