forkineye/ESPixelStick #710: ESP Crashes when asked to play a compressed file via the UI.

MartinMueller2003, 2023-12-15:
### Discussed in https://github.com/forkineye/ESPixelStick/discussions/706

<div type='discussions-op-text'>

<sup>Originally posted by **GeoTime61** December 12, 2023</sup>
I am trying to run a DigQuad ESP32 controller with an SD card as an FPP Remote. From xLights I can upload the controller configuration and use FPP Connect to upload sequence files. The files show up on the DigQuad and FPP sees the DigQuad as a Remote device. However, when FPP plays a sequence the LEDs do not turn on. If I take the DigQuad out of remote mode and send DDP data to it directly from FPP everything works so I think it is hooked up correctly. I am using FPP version 7.4 and ESPixelStick version 4.0-beta5. Has anyone had success with a DigQuad or other ESP32 controller using FPP Remote?</div>

kingofmonkeys, 2023-12-19:
I’ve been fighting with something similar or maybe the same issue.  I cant get local sequence files to play in my esp32s but everything works on my 8266s. Troubleshooting last night and i do appear to get what looks like a crash in serial output but im also seeing a message about start channel * frames not being right.  I ran out of time to troubleshoot but was planning to do some clean builds of the latest code to try again. 

Are you seeing errors in the serial output?

MartinMueller2003, 2023-12-19:
That is a malformed fseq file. It usually means the file has been truncated or has had data added to the end. It also happens when you configure fewer outputs on the EPS than are required to process the fseq file.

kingofmonkeys, 2023-12-19:
That makes sense for that error. Could this also cause the controller to crash?

MartinMueller2003, 2023-12-19:
Yes and that has been fixed. Latest version in main no longer crashes on bad files.

closed 2023-12-20 by MartinMueller2003