forkineye/ESPixelStick #720: ESPixelStick-4.0-beta5 - Output issue WS2811

Noelle88, 2023-12-22:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
ESPixelStick-4.0-beta5

**Hardware Version**
ESP32 DEV

**Binary release or compiled yourself?**
Binary (Generic ESP32 4MB)

**Operating System (and version)**
Win10

**Web Browser (and version)**
Firefox 121.0 (64-Bit)

**Access Point**

**Describe the bug**
1. With 4.0-beta 4 it works well, apart from the known bugs.

2. With 4.0-Beta 5 the outputs do not work correctly with exactly the same settings.

I have 2 universes with 471 channels each via E1.31.
Channel 1 and 2 each have 157 pixels and should actually be WS2811 pixels.

With beta 5 I get the data displayed in the diagnostics when I send the internal effects or external data,
but the pixels don't show anything.
However, they are turned off when you turn on the controller. The outputs appear to be active.

The "solution" of the problem:
Out of frustration I chose USC1903 instead of WS2811 and it works now?!
Very confusing and I think there is a mistake in the code ;-)

MartinMueller2003, 2023-12-26:
Cant reproduce this. My WS2811 pixels are working fine.

Noelle88, 2023-12-27:
Okay, thanks for testing. 
I will try again with another controller and browser in a few days.

MartinMueller2003, 2024-02-27:
Any update on this?

MartinMueller2003, 2024-03-29:
Closing this issue since I cannot reproduce and the OP has not been able to provide additional information.

closed 2024-03-29 by MartinMueller2003