forkineye/ESPixelStick #724: APA102 Support

marcobrianza, 2024-01-05:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
ESPixelStick_Firmware-4.0-ci7390878331

**Hardware Version**
Lolin D32 PRO

**Binary release or compiled yourself?**
CI release

**Describe the bug**

I was expecting support for APA102 since it is an ESP32 board but could not find the option as output in the combo box

MartinMueller2003, 2024-01-05:
APA102 is a compile time option. There was not a lot of call for it and I needed space for other protocols. You can recompile a version with it turned on.

marcobrianza, 2024-01-06:
thank you for the clarification I'll try to rebuild the software, which I never attempted due to the extensive board support already present. Since there is a space problem I guess I need to disable other protocols. I'll go happily without Renard and GECE which to me are more exotic than a classic APA102.

marcobrianza, 2024-01-06:
Hi @MartinMueller2003  I rebuilt the lastest code but I'm not sure where to find the option to enable APA102

MartinMueller2003, 2024-01-06:
The magic happens in the GPIO Def files. There is one for each platform. Go to your platform and uncomment the APA102 line and comment out something else. For APA you will also need to set the SPI GPIO values to what you are using.

marcobrianza, 2024-01-06:
I think that magic was already there for the D32PRO am I wrong?

MartinMueller2003, 2024-01-06:
Just loaded up my pro with latest code. You will find APA102 and WS2801 on Output port 7

MartinMueller2003, 2024-01-06:
That device has PSRAM and that allowed me to have all protocols enabled.

marcobrianza, 2024-01-06:
ok, very good I see it. before I did not explore all the ports to look for specific protocols

marcobrianza, 2024-01-06:
thank you for going through loading the code on the board to help me!

MartinMueller2003, 2024-01-06:
My pleasure

marcobrianza, 2024-01-06:
I'm Back :-)

setting output 7 to APA102 (that by default uses GPIO 16 and 17) in D32PRO
https://github.com/forkineye/ESPixelStick/blob/6181dd9cb37ec9d4d48e89f007b6af4edd00d3e4/ESPixelStick/src/platformDefinitions/GPIO_Defs_ESP32_LoLin_D32_PRO.hpp#L32
makes the board loop continuously

those GPIO are not availabe on D32 PRO for a reason and I think that is this:
https://github.com/micropython/micropython/issues/7819

I changed to these free ones

#define DEFAULT_SPI_DATA_GPIO   gpio_num_t::GPIO_NUM_15
#define DEFAULT_SPI_CLOCK_GPIO  gpio_num_t::GPIO_NUM_0

some thing happens to the LED but it does not look right to me. maybe I need to use other pins, what are the pins you recommend?

MartinMueller2003, 2024-01-06:
Make sure the pins you use are not used by any other output ports. Tyr GPIO 14 & 12

MartinMueller2003, 2024-01-06:
FYI: I do not have any APA102 pixels. Someone else tested them a long time ago. I would not be surprised if there is a timing issue.

marcobrianza, 2024-01-07:
I have been investigating the APA102 support further but cannot make it work. I have tried GPIO 14 and 12 without success.
To be sure it was not a hardware issue I tested WLED and the good old FastLED with success on the mentioned pins. They work flawlessly.
I had a look at the code but sort of got lost in it. When I stated using APA102 10 years ago I had this reference
https://cpldcpu.wordpress.com/2014/11/30/understanding-the-apa102-superled/
In that era I was using AVR with hardware SPI writing frames in a very simple way with no timing issues due to the clocked nature of the protocol. (I can share that code if needed)
For me trying to port things in project is out of my knowledge but I can support it testing and contribute donating a APA102 strip to the project.
Please tell me if I can contribute in the above ways.
I relay like the scope of this software focusing on interfacing LAN protocols to LED protocols, the ability to play sequences and the clean and logic UI.

MartinMueller2003, 2024-01-07:
I dont need a lot of pixels for testing. 10 or 20 pixels would be enough. Please contact me at MartinMueller2003@yahoo.com and I will give you my address.

MartinMueller2003, 2024-01-11:
Fixes have been uploaded to the latest PR.

marcobrianza, 2024-01-22:
I confirm the fixes are working

closed 2024-01-22 by marcobrianza

MartinMueller2003, 2024-01-22:
And my APA102 string just arrived in my mailbox today. Sigh.