forkineye/ESPixelStick #731: SD Card not installed

thughes2001, 2024-01-17:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0 beta 5

**Hardware Version**
Quinled Uno AE+

**Binary release or compiled yourself?**

**Operating System (and version)**
Window 10

**Web Browser (and version)**

**Access Point**

**Describe the bug**
Loading firmware and serial output shows  the SD card size and even files on the SD card then comes up with card not installed. How can that be?

22:22:05: [FileMgr] SD Card Size: 7580MB
22:22:08: [FileMgr] > System Volume Information
22:22:08: [FileMgr]   WPSettings.dat - 12
22:22:08: [FileMgr]   IndexerVolumeGuid - 76
22:22:08: [FileMgr] 12 Days.fseq - 3031537
22:22:08: [FileMgr] Linus Christmas Speech.fseq - 6035
22:22:08: [FileMgr] The_First_Noel.fseq - 113547
22:22:08: [FileMgr] Alvin 40fps.fseq - 953129
22:22:08: [FileMgr] Bob Seger Little Drummer boy.fseq - 40565
22:22:08: [FileMgr] Christmas Eve-Sarajevo.fseq - 709293
22:22:08: [FileMgr] Frosty.fseq - 82448
22:22:08: [FileMgr] Here Comes Santa.fseq - 263662
22:22:08: [FileMgr] It is begining to look like Christmas.fseq - 253033
22:22:08: [FileMgr] Configuration File: '/config.json' loaded.
22:22:08: [  6480][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
22:22:11: [  6980][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
22:22:11: [FileMgr] No SD card installed

thughes2001, 2024-01-17:
If the card is not installed, how can the firmware read it then not?

MartinMueller2003, 2024-01-18:
There has been a seperate conversation on this. It looks like the problem is the code starts to read the SD card. The low level driver encounters an error with the data coming from the SD card and gives up. There is nothing the V4 code can do about this. I know I am using the same version and have now tested as many as 75 files on the SD card without issues. 

I suspect there is a file with some odd encoding in it that causes an issue.
Please create an image file of your SD card and send it to me so I can install it and see what is happening.

thughes2001, 2024-01-18:
I would love to send the image but having a problem with the size of the image file. Tried Google sharing, tried Dockbox. No luck with either. Dockbax said 20hrs to upload.
Tim Hughes

thughes2001, 2024-01-18:
Is there a preferred Micro SD card to use with EXPixelStick when using SD card?Tim Hughes

MartinMueller2003, 2024-01-18:
Most of my SD cards are Sandisk Ultra 16GB. 
Can you send me the three files before it stops reading and the next three files it would read?

thughes2001, 2024-01-18:
Not sure how to get the files you are asking for since the flash is a Bin if I am right.
this is the serial output.
15:45:53: [FileMgr] 'ws2801.html.gz':     '1492'
15:45:53: [FileMgr] 'ws2811.html.gz':     '1577'
15:45:53: [FileMgr] SD Card Size: 7580MB
15:45:53: [FileMgr] > System Volume Information
15:45:53: [FileMgr]   WPSettings.dat - 12
15:45:53: [FileMgr]   IndexerVolumeGuid - 76
15:45:53: [FileMgr] test.txt - 1048576
15:45:53: [FileMgr] foo.txt - 13
15:45:53: [FileMgr] hello.txt - 13
15:45:53: [FileMgr] > test
15:45:53: [FileMgr] > FOUND.000
15:45:53: [FileMgr]   FILE0000.CHK - 32768
15:45:53: [FileMgr] Configuration File: '/config.json' loaded.
15:45:54: [  3576][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
15:45:56: [  4076][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
15:45:57: [FileMgr] No SD card installed
15:45:57: [FileMgr] Configuration File: '/input_config.json' loaded.
15:45:57: [FileMgr] Configuration File: '/output_config.json' loaded.
15:45:57: [FileMgr] Configuration File: '/admininfo.json' saved 176 bytes.
15:45:57: [FileMgr] Configuration File: '/config.json' saved 387 bytes.
From what I am seeing, it it trying to use CS @15:45:56? Is it not seeing GPIO5 on the ESP32 used for SD CS. Also the card is an 8GB.
I tried to upload the disk image to DropBox but that kept locking up. Doesn't like the size of the file.

Tim Hughes

MartinMueller2003, 2024-01-18:
Pull the sdcard out of the ESP. Put it in your PC. copy the last displayed file and the first not displayed file into a zip and send it to me. Also, the only files that should be present on the SD card are fseq files.

thughes2001, 2024-01-18:
Thank you for trying to help. Just don't understand what you are asking for or how to get it.
I put a SD card in the slot on the module, connect to PC with USB.Open ESpixelStick Flash Tool.Set up the SSIDSet the static IPSet gateway to my routerUnder Hardware select 
QuinLed Uno AE+Insure the com portthen Flash the Device.It goes through System Output
-= Uploading Firmware =-
Command:  bin/win32/python3/python bin/upload.py --chip esp32 --baud 460800 --port COM7 --before default_reset --after hard_reset write_flash -z 0x1000 firmware/esp32/esp32_quinled_uno_ae_plus-bootloader.bin 0x8000 firmware/esp32/esp32_quinled_uno_ae_plus-partitions.bin 0xe000 firmware/esp32/boot_app0.bin 0x10000 firmware/esp32/esp32_quinled_uno_ae_plus-app.bin 0x3B0000 firmware/filesystem.bin
esptool.py v2.8
Serial port COM7
Connecting........_
Chip is ESP32D0WDQ5 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:d1:f9:5b:13:94
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 17088 bytes to 11848...

Writing at 0x00001000... (100 %)
Wrote 17088 bytes (11848 compressed) at 0x00001000 in 0.3 seconds (effective 424.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 391.1 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 1084.6 kbit/s)...
Hash of data verified.
Compressed 1108560 bytes to 687443...

Writing at 0x00010000... (2 %)
Writing at 0x00014000... (4 %)
Writing at 0x00018000... (7 %)
Writing at 0x0001c000... (9 %)
Writing at 0x00020000... (11 %)
Writing at 0x00024000... (14 %)
Writing at 0x00028000... (16 %)
Writing at 0x0002c000... (19 %)
Writing at 0x00030000... (21 %)
Writing at 0x00034000... (23 %)
Writing at 0x00038000... (26 %)
Writing at 0x0003c000... (28 %)
Writing at 0x00040000... (30 %)
Writing at 0x00044000... (33 %)
Writing at 0x00048000... (35 %)
Writing at 0x0004c000... (38 %)
Writing at 0x00050000... (40 %)
Writing at 0x00054000... (42 %)
Writing at 0x00058000... (45 %)
Writing at 0x0005c000... (47 %)
Writing at 0x00060000... (50 %)
Writing at 0x00064000... (52 %)
Writing at 0x00068000... (54 %)
Writing at 0x0006c000... (57 %)
Writing at 0x00070000... (59 %)
Writing at 0x00074000... (61 %)
Writing at 0x00078000... (64 %)
Writing at 0x0007c000... (66 %)
Writing at 0x00080000... (69 %)
Writing at 0x00084000... (71 %)
Writing at 0x00088000... (73 %)
Writing at 0x0008c000... (76 %)
Writing at 0x00090000... (78 %)
Writing at 0x00094000... (80 %)
[... 149 lines cut ...]
16:50:03: [FileMgr] Here Comes Santa.fseq - 263662
16:50:03: [FileMgr] It is begining to look like Christmas.fseq - 253033
16:50:03: [FileMgr] Configuration File: '/config.json' loaded.
16:50:03: [  3666][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
16:50:06: [  4166][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
16:50:06: [FileMgr] No SD card installed
16:50:06: [FileMgr] Configuration File: '/input_config.json' loaded.
16:50:06: [FileMgr] Configuration File: '/output_config.json' loaded.
16:50:07: [FileMgr] Configuration File: '/admininfo.json' saved 176 bytes.
16:50:07: [FileMgr] Configuration File: '/config.json' saved 391 bytes.
16:50:07: [WiFiDrv] WiFi Entering State: Connecting Using Config Credentials
16:50:08: [WiFiDrv] Using Static IP
16:50:08: [WiFiDrv] Connecting to 'shadow' as esps-94135BF9D108
16:50:08: [WiFiDrv] WiFi Entering State: Connected To AP
16:50:09: [WiFiDrv] Connected with IP: 192.168.0.206
16:50:09: [WebMgr] Web server listening on port 80
16:50:09: [  FPPD] FPPDiscovery subscribed to broadcast messages on port: 32320
16:50:09: [  FPPD] FPPDiscovery subscribed to multicast: 239.70.80.80:32320
16:50:09: 

FileMGR
Then goes to WifiDrvthen WebMgrthen FPPD
Thank for all you tried. Maybe This is above my head.I did get the module to show in Diagnostic RGB when driven by a Kulp K16A as the Player with a show upload to the Kulp.I guess that is a win. Just no SD. Was hoping to use the module as a Remote device instead of the Kulp sending all the data.I guess I can use it an ESPixelStick or Wled UNO. they will do the same thing but WLed I can select the pin for Port output.

Thanks Tim

MartinMueller2003, 2024-01-18:
It looks like you are loading the beta 5 version. Please use the dist file found here:

https://github.com/MartinMueller2003/ESPixelStick/releases

thughes2001, 2024-01-18:
Thank You Very MuchThat seems to have worked.I now see File manager and files on the SD when connect to the module through XLightsAgain, Thanks
Tim Hughes

 
 

https://github.com/MartinMueller2003/ESPixelStick/releases

thughes2001, 2024-01-18:
Just wondering. If you plan on using the ESPixelStick as a remote controller, you would disable Primary Input and configure Secondary Input to FPP Remote. Am I wrong that if the primary is set, ESPixelStick would look for all data from Player and not call for the data on the SDCard?
Am I wrong on this. It seems to work with Primary disables and secondary set to FPP Remote and shows up in XLight tool, FPP Connect.
Thank you VERY MUCHIt seems to work now

Tim Hughes

 
 

https://github.com/MartinMueller2003/ESPixelStick/releases

MartinMueller2003, 2024-01-19:
If primary is not receiving data, then the ESP uses the secondary input. It is possible to have both set up, use ddp to play something special and then revert to secondary to play an fseq file. People also do this to test. Send test patterns using ddp and when you stop the device syncs to the show. As you have found, you can also leave the primary input disabled.

closed 2024-01-19 by MartinMueller2003