forkineye/ESPixelStick #732: ESPixelStick-4.0-beta5 Multi LED Protocol Output Bug

Hedgehog-Info, 2024-01-17:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-beta5

**Hardware Version**
sp32_quinled_dig_octa

**Binary release or compiled yourself?**
Binary

**Operating System (and version)**
Win11

**Web Browser (and version)**
Chrome Version 120.0.6099.224 (Official Build) (64-bit)

**Access Point**
?? Not sure what this means

**Describe the bug**
Hi,  I'm using the SD card to play a remote sequence from xLights and it works fine if all device output settings have the same protocol.  (WS2811 for example)  There is a an error that crashes the board and resets the output configuration to "disabled"

Steps to reproduce:

1. Create a xLights sequence with WS2811 protocol on port 1 and ports 2 and 3 with the UCS1903 protocol.
2. Upload the file to the memory card on the board
3. Go to the webportal and click on device setup.  Configure to play remote sequence
4. Configure ports/outputs to according to your xLights sequence (Mixture of WS2811 and USC1903).
5. Hit save (Crash occurs here)
6. Webserver will restart but all output settings return to "disable"

Another thing to note is that the file management tab... The date/time of the uploaded file is not correct...  

[Screenshots.pdf](https://github.com/forkineye/ESPixelStick/files/13969798/Screenshots.pdf)

MartinMueller2003, 2024-01-18:
I just configured WS2811 on port 1, UCS1903 on ports 5 & 6 (8 port device)  from a clean start and did not have a crash and it is working fine. 

NOTE: A change in protocol on a port requires a reboot since the IDF drivers do not properly support disconnecting an output. So if you changed from DMX to UCS1903 on a port then there is an expected reboot. 

After the reboot it should come up in the desired mode. I did not see it come up in the disabled mode after a protocol was selected. 

Please post a screenshot of the admin page. Please ensure you are running the latest dist found here: https://github.com/MartinMueller2003/ESPixelStick/releases

Hedgehog-Info, 2024-01-18:
@MartinMueller2003 thank you for looking into this.  I changed my steps based on what you suggested and it appears to be more stable.

1. Create a xLights sequence with WS2811 protocol on port 1 and ports 2 and 3 with the UCS1903 protocol.
2. Upload the file to the memory card on the board
3. Go to the webportal and click on device setup. Configure to play remote sequence
4. Configure a **SINGLE** output hit "Save"
5. Go to the admin Panel  and hit "Reboot"
6. Repeat steps 4-5 until all the outputs are configured.  (If I configure to many at a time and seems cash the issue as stated above)

MartinMueller2003, 2024-01-18:
That is not quite what I meant. :)

I just configured 5 outputs at the same time (single save) and it did not crash. It WILL reboot (not a crash) if you change the output type after an output has been configured. This is needed due to limitations of the RMT low level driver.

MartinMueller2003, 2024-01-18:
[image]

Hedgehog-Info, 2024-01-18:
https://github.com/forkineye/ESPixelStick/assets/156968310/c436acec-125b-44ba-adc3-aadec41645f5

Thank you.  I really appreciate the support and knowledge.  Yes gotcha but you got me thinking about performing more saves.    I have uploaded a video of this happening.  40s-60s you can see that I have defined multiple outputs and then around 55s everything resets.    Another thing that I just noticed is that the pixel count defaults to 100 when issue occurs.    If it defaults to the proper count then everything seems to work?

MartinMueller2003, 2024-01-20:
That build is not the latest build and a lot of issues have been fixed since then. Please use the newest dist.zip from https://github.com/MartinMueller2003/ESPixelStick/releases

Hedgehog-Info, 2024-01-20:
Thank you again.  Downloaded the last dist.zip and it appears more stable.  =)    I have taken some screenshots what I'm now witnessing...   

Steps to reproduce:

1) Create a very basic xlights file with multiple outputs.  WS2811 on output 1 and UCS1903 on ports 2-5
2) Upload file and set all the outputs correctly.  Verify output works.  This seemed to work much better!
3) Re-Upload the same file.  What happens is that the output freezes and ports 2-5 are reset to disabled..  All outputs are hung based on what was playing prior to the file upload.  This can be seen by the screenshots below.

Only the first 25 lights (output 1) are working and changing color..  After the 25th box are all frozen.   
[image]

Notice outputs 2-5 are reset to disabled.  This matches with the diagnostics above of being frozen.
[image]

**Other screenshots:**
[image]
[image]

Hedgehog-Info, 2024-02-27:
> I have tried multiple times to recreate this and it does not happen on my
> card. Not sure what to do next. It looks to me as if you are having a file
> corruption issue. That is in the framework code. Not easy to debug.

No problem.  If other people have similar issues LMK and I can troubleshoot
it with you real time.  Like I said I can make it work but takes brute
force but once the settings stick it seems stable for at least 4hrs.  (The
length of my test)

Thanks again

MartinMueller2003, 2024-02-27:
No one else is reporting such an issue.

closed 2024-05-16 by MartinMueller2003