forkineye/ESPixelStick #743: Can't connect

philspitlerSF, 2024-03-02:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
3.2

**Hardware Version**

**Binary release or compiled yourself?**
Binary release

**Operating System (and version)**
OSX or Windows

**Web Browser (and version)**
Chrome or Safari

**Access Point**
Home network not AP mode

**Describe the bug**
The device is running my pixels ok but most of the time I can't get to the UI in my browser.
In fact, Chrome and Safari try and download a file when I go to the IP of the PixelStick.
If I power cycle the board I can usually connect again.
The network settings are set to connect to my home network and I can ping the ESP.

MartinMueller2003, 2024-03-02:
Not much development is happening with V3. Have you tried V4 Software? You can get the latest dist file here:
https://github.com/MartinMueller2003/ESPixelStick/releases/download/dist/dist.zip

philspitlerSF, 2024-03-02:
I tried updating using Windows and OSX and get this error.

I don't recall having to set it into a mode before.

Any ideas?

[image]

philspitlerSF, 2024-03-02:
There is an option to update the firmware in the UI but I'm unsure which file to choose from the repo as non of them become options when I browse the folders.

MartinMueller2003, 2024-03-10:
Lost track of this conversation. Using the latest dist.zip file found here: https://github.com/MartinMueller2003/ESPixelStick/releases/download/dist/dist.zip

If you have true ESPS V3 HW then in the flashtool you select ESPSv3 as the software version and make sure the com port is set properly. Then press upload.

iz1kbp, 2024-03-27:
Hi Martin, I've the same problem: I'm not able to program the unit.
I use wemos d1 mini. I've try to program my board with arduino IDE and all is right,but with you JAR (v3.2 and 4.0 too) I'm not able to finalize programm proces on  mywemos d1 mini.
I'm using w7/32
[image]

[log.txt](https://github.com/forkineye/ESPixelStick/files/14775734/log.txt)

thank you for help!
Mario

MartinMueller2003, 2024-03-27:
COM20, I have never seen a number that high.
With the flashtool turned on, and the com port set and the comm cable connected, if you reset the d1Mini, what comes out on the serial log window? You should see the default booting string. If not, then there is a cable or setting issue. Even if garbage characters come out that is enough to prove the cable is working.

iz1kbp, 2024-03-27:
Hi martin.I've reistall Java (last rev from your link) and usb driver (from your link).I'change the usb port on my pc.
Now I'm on a com 8
but...same problem
I've try to upload a simple schetch from IDE withou problem. (cattura.png)
[image]
[image]

iz1kbp, 2024-03-27:
[log.txt](https://github.com/forkineye/ESPixelStick/files/14776573/log.txt)

iz1kbp, 2024-03-27:
When I press reset on midi d1 I read somethin in com monitor
[image]

iz1kbp, 2024-03-27:
[image]
this is serial monitor in arduino IDE pressing RESET on D1 ... Note com speed

MartinMueller2003, 2024-03-27:
The odd character output is not unexpected. It does indicate a good cable and com port setting. Are you running on a mac or a windows system?

MartinMueller2003, 2024-03-27:
I just downloaded the latest dist file onto a windows machine and uploaded an image. Here is my log.
[log.txt](https://github.com/forkineye/ESPixelStick/files/14776749/log.txt)

iz1kbp, 2024-03-27:
I'm running on win7/32. Last java downloaded.

MartinMueller2003, 2024-03-27:
Do you have a more recent Windows version available? I never tested that far back.

iz1kbp, 2024-03-27:
Mmmm...no, I use w7 to develop and program Arduino. Are the old(3.2 or minor) revision ok on W7 system?

MartinMueller2003, 2024-03-27:
I do not know why it is not working on your system. It should as long as python and java are both the latest version.

iz1kbp, 2024-03-27:
Java is the last but I've no python installed,not directly...however.Should I install pyuthon?

iz1kbp, 2024-03-27:
Ok Martin....under win7 the program is not working.I've installed (light install) win10 on another pc and all work's fine. Only a question: is the system decode only sacn (i've read 1.31 in set up,not artnet) or sacn and artnet too? the out data is the tx pin of the wemos d1?
Thank again.

Mario

MartinMueller2003, 2024-03-27:
Not sure what you mean by system decode. We accept E1.31 sACN artnet ddp as possible input protocols. We also support fseq file format to play files locally.

iz1kbp, 2024-03-28:
Hi Martin, what I mean is this: in the stable version 3.2 the artnet protocol is not specified but only 1.31 (sacn). In fact I programmed D1 with version 3.2 and it only decodes sacn messages. Programming D1 with version 4 (dist ) I have the option to choose Artnet or SACN. I did some tests and in fact, by programming D1 with the DIST version, I can decode both Artnet and sACN. I noticed a problem: in ARTNET, if I set the input universe to ZERO (0) and transmit on ZERO universe, no decode nothing. If I transmit on ONE and set D1 on universe 1 the operation is OK.
Technical INFO. I already use Wemos d1 mini with my Arduino program in which I have directed the DMX output to the serial TX (tx >rs485). Does your program send the data to the TX pin of wemos d1? If so, for me it's just a question of updating the software without having to modify my hardware.

MartinMueller2003, 2024-03-28:
3.2 is no longer under active development. I was describing what V4 can do. V4 is a superset of V3. I will check to see if universe zero is a valid setting (I think it is) and see what is not working. If you set the host name properly in the network config, then you should be able to address the ESP using the host name. Sometimes your AP will fail to register the hostname with the local DNS entity.

iz1kbp, 2024-03-28:
Hi Martin. I meant if it responds to the ArtNet Poll command, not to the browser search. ArtNet Poll provides an OpCode sent via the network to which the devices respond with strings called ShortName and Large name

MartinMueller2003, 2024-06-25:
It looks like the initial issue is resolved.