forkineye/ESPixelStick #745: WebUI Update (EFU file)

philspitlerSF, 2024-03-04:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.x

**Hardware Version**
d1 Mini

**Binary release or compiled yourself?**
Binary release

**Operating System (and version)**
OSX 14.3

**Web Browser (and version)**
Chrome 

**Access Point**
Nope

**Describe the bug**
When I try an update the firmware in the webUI, it is looking for an EFU file but I don't see any files with that extension.

https://github.com/forkineye/ESPixelStick/assets/15186070/a3810ba6-2f43-44b1-9ea1-76cec790ef9f

MartinMueller2003, 2024-03-04:
You need to create the EFU using the flashtool.

philspitlerSF, 2024-03-04:
Great thanks. I did that and now it says it's updating but the progress bar hasn't moved in 10 mins.

MartinMueller2003, 2024-03-04:
I have not tried it in over a year. It is possible you have hit a bug. Almost everyone uses the flashtool to do updates.

MartinMueller2003, 2024-03-04:
That version is your picture is WAY old. I hope you are updating to the version in the dist that I sent the look for.

forkineye, 2024-03-04:
EFU support doesn't work in the older 4.x beta / CI versions.

philspitlerSF, 2024-03-04:
Thanks.. I managed to update via the Terminal so all good now. Thanks

MartinMueller2003, 2024-03-04:
I am not near any equipment for a few days. Are you saying EFU update on current CI build is broken?

philspitlerSF, 2024-03-04:
No, it was a much older build.

MartinMueller2003, 2024-03-05:
It looks like we get part way through and then the browser stops sending data. Last transaction stats info

------ FirmwareUpload:ESPixelStick/src/WebMgr.cpp:713:  file: 'ESP32 TTGO T8.efu' ------
------ FirmwareUpload:ESPixelStick/src/WebMgr.cpp:714: index: 132911 ------
------ FirmwareUpload:ESPixelStick/src/WebMgr.cpp:715:  data: 0x3ffdcbf8 ------
------ FirmwareUpload:ESPixelStick/src/WebMgr.cpp:716:   len: 1436 ------
------ FirmwareUpload:ESPixelStick/src/WebMgr.cpp:717: final: 0 ------
------ FirmwareUpload:ESPixelStick/src/WebMgr.cpp:729: In AP Mode ------
------ FirmwareUpload:ESPixelStick/src/WebMgr.cpp:744: Sending data to efupdate ------
------ FirmwareUpload:ESPixelStick/src/WebMgr.cpp:750: Packet has been processed ------
------ FirmwareUpload:ESPixelStick/src/WebMgr.cpp:758: No EFUpdate Error ------
------ FirmwareUpload:ESPixelStick/src/WebMgr.cpp:770: End ------

MartinMueller2003, 2024-03-07:
This is a multi bug issue.
First bug was a pair of checks that were guaranteed to reject an upload request. First check only allowed uploads in AP mode. 2nd check only allowed uploads in STA mode. fail either and your request gets canceled. Next bug is on the UI for not telling you the request was rejected and just sitting there with a progress indicator.

First issue has been fixed. UI issue is still open. 
Testing repeatability and about 50% of the transfers hang part way through. The transfers that fail while the FS is being updated tend to make the device unusable until you use the flashtool to upload the images.

MartinMueller2003, 2024-03-07:
making the writes take longer causes the hangs to happen sooner and more often. Not saving (as in zero time used) still result in 50% hangs around 400000 bytes "Saved". So at this time, if it is a time related issue, the bulk of the problem is in the ESPAsyncWebServer processing.

MartinMueller2003, 2024-03-10:
Had to update the platform code to get the reliability up. Tested 25/25 with no failures. PR request has been created and dist.zip has been updated.

closed 2024-03-10 by MartinMueller2003