forkineye/ESPixelStick #750: restore configuration issue (on dig octa)

marcobrianza, 2024-03-11:
**ESPixelStick Firmware Version**
4.0-ci8237205806

**Hardware Version**
dig octa

**Binary release or compiled yourself?**
ci release

**Describe the bug**
restoring the configuration produces an error and parameters are not applied
I saved this file

[bubbleid-4C7F5A1B5AE0-network.json](https://github.com/forkineye/ESPixelStick/files/14563253/bubbleid-4C7F5A1B5AE0-network.json)

applied a factory reset

loading the file produces this log:

19:34:51.164 -> [FileMgr] Configuration File: '/config.json' saved 1436 bytes.
19:34:51.209 -> [FileMgr] Configuration File: '/config.json' saved 1436 bytes.
19:34:51.307 -> [FileMgr] Configuration File: '/config.json' saved 1436 bytes.
19:34:51.383 -> [FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
19:34:51.489 -> [FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
19:34:51.526 -> [FileMgr] Configuration File: '/config.json' saved 1436 bytes.
19:34:51.636 -> [FileMgr] Configuration File: '/config.json' saved 1436 bytes.
19:34:51.781 -> [FileMgr] Configuration File: '/config.json' saved 1436 bytes.
19:34:51.890 -> [FileMgr] Configuration File: '/config.json' saved 1436 bytes.
19:34:51.997 -> [FileMgr] Configuration File: '/input_config.json' saved 979 bytes.
19:34:52.182 -> [FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
19:34:52.691 -> [FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
19:34:52.764 -> [FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
19:34:52.876 -> [FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
19:34:52.947 -> [FileMgr] Configuration File: '/config.json' saved 1436 bytes.
19:34:52.981 -> [FileMgr] Configuration File: '/input_config.json' saved 1010 bytes.
19:34:53.095 -> [FileMgr] Configuration File: '/config.json' saved 1436 bytes.
19:34:53.164 -> [FileMgr] Configuration File: '/config.json' saved 1343 bytes.
19:34:53.273 -> [FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
19:34:53.349 -> [FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
19:34:53.562 -> [FileMgr] Configuration File: '/output_config.json' saved 143 bytes.
19:34:57.327 -> [FileMgr] Configuration File: '/input_config.json' loaded.
19:34:57.690 -> [FileMgr] Configuration File: '/config.json' loaded.
19:34:57.690 -> [   ESP] Missing Config Version ID
19:34:57.690 -> [   ESP] No device settings found.
19:34:57.728 -> [FileMgr] No File Manager settings found.
19:34:57.728 -> [NetworkMgr] No network config found. Use default settings
19:34:57.799 -> [FileMgr] Configuration File: '/config.json' saved 634 bytes.
19:34:58.503 -> [FileMgr] Configuration File: '/output_config.json' loaded.

MartinMueller2003, 2024-03-12:
GPIO-1 and GPIO-3 are the serial console pins

closed 2024-03-14 by MartinMueller2003

MartinMueller2003, 2024-03-15:
Also made an attempt to merge the new and old config to create fields for features that may have been added since the backup was made.

marcobrianza, 2024-03-17:
Hi I was testing the save restore feature from scratch with the latest CI and found some issues.
The network settings look ok but the "Device Setup" is not, looking in the saved config I think it might be an issue in the backup function.

MartinMueller2003, 2024-03-17:
I just did a save and the device config is certainly there. Keep in mind that the data in the config file does NOT strictly follow the layout you find on the UI. Fields are located to make the code work better, not make it easy on humans.
{
    "system": {
        "cfgver": 1,
        "device": {
            "id": "ESPixelStick",
            "blanktime": 5,
            "miso_pin": 36,
            "mosi_pin": 15,
            "clock_pin": 14,
            "cs_pin": 16
        },
        "network": {
            "hostname": "esps-502F3FEF49C0",
            "wifi": {
                "ssid": "",
                "passphrase": "",
                "ap_ssid": "ESPixelStick-AP-esps-502F3FEF49C0",
                "ap_passphrase": "",
                "ip": "0.0.0.0",
                "netmask": "0.0.0.0",
                "gateway": "0.0.0.0",
                "StayInApMode": false,
                "dhcp": true,
                "sta_timeout": 15,
                "ap_channel": 1,
                "ap_fallback": true,
                "ap_timeout": 120,
                "ap_reboot": false
            },
            "weus": true,
            "eth": {
                "ip": "192.168.10.223",
                "netmask": "255.255.255.0",
                "gateway": "192.168.10.1",
                "dhcp": false,
                "type": 0,
                "addr": 0,
                "power_pin": -1,
                "mode": 3,
                "mdc_pin": 23,
                "mdio_pin": 18,
                "activevalue": false,
                "activedelay": 1000
            }
        }
    },
    "input": {
        "cfgver": 1,
        "ecb": {
            "enabled": false,
            "name": "Effects Control",
            "id": 0,
            "polarity": "ActiveLow",
            "channels": 32,
            "long": 2000
        },
        "channels": {
            "0": {
                "0": {
                    "type": "E1.31",
                    "universe": 1,
                    "universe_limit": 512,
                    "universe_start": 1,
                    "port": 5568
                },
                "4": {
                    "type": "DDP"
                },
                "6": {
[... 1123 lines cut ...]
                            "b16": false,
                            "sca": true,
                            "hv": 0
                        },
                        {
                            "id": 13,
                            "en": false,
                            "Min": 650,
                            "Max": 2350,
                            "rev": false,
                            "b16": false,
                            "sca": true,
                            "hv": 0
                        },
                        {
                            "id": 14,
                            "en": false
                        }
                    ]
                },
                "type": 0
            }
        }
    }
}

marcobrianza, 2024-03-17:
on Dig-Octa this is whay I see:

[image]

this is the file
[bubble-5C695A1B5AE0-01.json](https://github.com/forkineye/ESPixelStick/files/14627735/bubble-5C695A1B5AE0-01.json)

I expect to find the sequence name Roma5.seq for example

aslo here:
{
	"system": {
		"cfgver": 1,
		"device": {
			"id": "bubble",
			"blanktime": 5,
			"miso_pin": 36,
			"mosi_pin": 15,
			"clock_pin": 14,
			"cs_pin": 16
		},
		"network": {
			"hostname": "bubble",
			"wifi": {
				"ssid": "",
				"passphrase": "",
				"ap_ssid": "bubble",
				"ap_passphrase": "bubble01",
				"ip": "0.0.0.0",
				"netmask": "0.0.0.0",
				"gateway": "0.0.0.0",
				"StayInApMode": true,
				"dhcp": true,
				"sta_timeout": 15,
				"ap_channel": 1,
				"ap_fallback": true,
				"ap_timeout": 120,
				"ap_reboot": false
			},
			"weus": true,
			"eth": {
				"ip": "192.168.72.90",
				"netmask": "255.255.255.0",
				"gateway": "192.168.72.254",
				"dhcp": false,
				"type": 0,
				"addr": 0,
				"power_pin": -1,
				"mode": 3,
				"mdc_pin": 23,
				"mdio_pin": 18,
				"activevalue": false,
				"activedelay": 1000
			}
		}
	},
	"input": {
		"cfgver": 1,
		"ecb": {
			"enabled": false,
			"name": "Effects Control",
			"id": "0",
			"polarity": "ActiveLow",
			"channels": 32,
			"long": "2000"
		},
		"channels": {
			"0": {
				"0": {
					"type": "E1.31",
[... 1119 lines cut ...]
							"id": 11,
							"en": false,
							"Min": 650,
							"Max": 2350,
							"rev": false,
							"b16": false,
							"sca": true,
							"hv": 0
						},
						{
							"id": 12,
							"en": false,
							"Min": 650,
							"Max": 2350,
							"rev": false,
							"b16": false,
							"sca": true
						}
					]
				},
				"type": 0
			}
		}
	}
}

marcobrianza, 2024-03-17:
restoring I get this

[image]

marcobrianza, 2024-03-18:
I tried again (latest release win dns) and can see errors in the restore log after reboot

[FileMgr] Configuration File: '/config.json' saved 748 bytes.
[FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
[FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
[FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
[FileMgr] Configuration File: '/input_config.json' saved 1436 bytes.
[FileMgr] Configuration File: '/input_config.json' saved 604 bytes.
[FileMgr] Configuration File: '/RestoredConfig.json' saved 23 bytes.
------ operator():ESPixelStick/src/WebMgr.cpp:334: Received RestoredConfig message ------
[FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
[FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
[FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
[FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
[FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
[FileMgr] Configuration File: '/output_config.json' saved 180 bytes.
[FileMgr] Configuration File: '/config.json' loaded.
[FileMgr] Configuration File: '/input_config.json' loaded.
[InputMgr] Shutting Down 'Disabled' on Input: 0
[InputMgr] Starting Artnet for channel '0'.
[Artnet] Subscribed to broadcast
[Artnet] Listening for 0 channels from Universe 1 to 0
[Artnet] Listening for 0 channels from Universe 1 to 0
[InputMgr] Shutting Down 'Disabled' on Input: 1
[InputMgr] Starting FPP Remote for channel '1'.
[FileMgr] Configuration File: '/output_config.json' loaded.
[OutputMgr]  Shutting Down 'Disabled' on Output: 0
[OutputMgr] 'WS2811' Initialization for Output: 0
[Artnet] Listening for 400 channels from Universe 1 to 1
[   ESP] ***-----Internal Reboot Requested. Rebooting Now
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (520) esp_core_dump_flash: No core dum⸮⸮⸮ѥѥ⸮⸮⸮found!
E (521) esp_core_dump_flash: No core dump partition found!

[   ESP] ESPixelStick v4.0-ci8317436315 (Mar 17 2024 - 18:25:30)
[   ESP] v4.4.6-dirty
[FileMgr] Flash file system initialized. Used = 225280 out of 327680
[FileMgr] Listing directory: /
[FileMgr] 'RestoredConfig.json': 	'23'
[FileMgr] 'UpdRecipe.json': 	'784'
[FileMgr] 'admininfo.json': 	'180'
[FileMgr] 'config.json': 	'748'
[FileMgr] 'input_config.json': 	'2040'
[FileMgr] 'output_config.json': 	'11668'
[FileMgr] Listing directory: //www
[FileMgr] 'alexa.html.gz': 	'146'
[FileMgr] 'apa102.html.gz': 	'1503'
[FileMgr] 'artnet.html.gz': 	'385'
[FileMgr] 'ddp.html.gz': 	'144'
[FileMgr] 'dmx.html.gz': 	'455'
[FileMgr] 'e1_31.html.gz': 	'460'
[FileMgr] 'effects.html.gz': 	'973'
[FileMgr] 'esps.css.gz': 	'13424'
[FileMgr] 'esps.js.gz': 	'62117'
[FileMgr] 'favicon.ico': 	'6450'
[FileMgr] 'fpp_remote.html.gz': 	'385'
[FileMgr] 'gece.html.gz': 	'501'
[FileMgr] 'gs8208.html.gz': 	'1579'
[FileMgr] 'index.html.gz': 	'5232'
[FileMgr] 'mqtt.html.gz': 	'712'
[FileMgr] 'relay.html.gz': 	'507'
[... 16 lines cut ...]
[FileMgr] Heap: 226956
**[FileMgr] Configuration File: '/output_config.json' Deserialzation Error. Error code = NoMemory**
[FileMgr] +++++{"output_config":{"cfgver":1,"MaxChannels":9000,"channels":{"0":{"0":{"type":"Disabled"},"1":{"type":"WS2811","color_order":"rgbw","pixel_count":"100","group_size":"1","zig_size":"1","gamma":"1","brightness":"100","interframetime":"300","prependnullcount":"0","appendnullcount":"0","data_pin":"0","showgamma":false},"2":{"type":"GECE","color_order":"rgb","pixel_count":63,"group_size":1,"zig_size":1,"gamma":1,"brightness":100,"interframetime":300,"prependnullcount":0,"appendnullcount":0,"data_pin":0},"8":{"type":"UCS1903","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":1,"gamma":1,"brightness":100,"interframetime":25,"prependnullcount":0,"appendnullcount":0,"data_pin":0},"9":{"type":"TM1814","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":1,"gamma":1,"brightness":100,"interframetime":500,"prependnullcount":0,"appendnullcount":0,"data_pin":0,"currentlimit":50},"12":{"type":"GS8208","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":1,"gamma":1,"brightness":100,"interframetime":300,"prependnullcount":0,"appendnullcount":0,"data_pin":0},"13":{"type":"UCS8903","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":1,"gamma":1,"brightness":100,"interframetime":30,"prependnullcount":0,"appendnullcount":0,"data_pin":0},"type":1},"1":{"0":{"type":"Disabled"},"1":{"type":"WS2811","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":1,"gamma":1,"brightness":100,"interframetime":300,"prependnullcount":0,"appendnullcount":0,"data_pin":1},"2":{"type":"GECE","color_order":"rgb","pixel_count":63,"group_size":1,"zig_size":1,"gamma":1,"brightness":100,"interframetime":300,"prependnullcount":0,"appendnullcount":0,"data_pin":1},"8":{"type":"UCS1903","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":1,"gamma":1,"brightness":100,"interframetime":25,"prependnullcount":0,"appendnullcount":0,"data_pin":1},"9":{"type":"TM1814","color_order":"rgb","pixel_count":100,"group_size":1,"zig_size":1,"gamma":1,"brightness":100,"interframetime":500,"prependnullcount":0,"appendnullcount":0,"data_pin":1,"currentlimit":50},"12":{"type":"GS8208","color_order":"rgb"," [... cut ...]
E (16548) gpio: gpio_set_level(227): GPIO output gpio_num error
[EthDrv] Entering State: Connecting
[EthDrv] ETH IP Before Start: 0.0.0.0
[EthDrv] Ethernet Connecting as esps-5C695A1B5AE0
[EthDrv] Connecting to Ethernet with Static IP
[EthDrv] Entering State: Wait for IP Address
[EthDrv] Entering State: Got IP
[WebMgr] Web server listening on port 80
[  FPPD] FPPDiscovery subscribed to broadcast messages on port: 32320
[  FPPD] FPPDiscovery subscribed to multicast: 239.70.80.80:32320
[EthDrv] Ethernet Connected with IP: 192.168.72.90

MartinMueller2003, 2024-03-18:
That happened once in my testing and then I could not get it to reproduce.

marcobrianza, 2024-03-18:
this very simple config file of a dig-octa can produce the issue on a rest device
[ESPixelStick-5C695A1B5AE0-t01.json](https://github.com/forkineye/ESPixelStick/files/14636072/ESPixelStick-5C695A1B5AE0-t01.json)

MartinMueller2003, 2024-03-18:
Fix has been uploaded and is part of the current PR. Octa has a huge output config and the json doc was not big enough to hold it under all configuration combinations and definitely not big enough to merge two configs together.

marcobrianza, 2024-03-18:
While I see no error, I still don't see the output configuration restored

marcobrianza, 2024-03-18:
after loading the file the output 1 is enabled correctly as by config but after reboot and merge is not

[
[ESPixelStick-5C695A1B5AE0-02.json](https://github.com/forkineye/ESPixelStick/files/14637893/ESPixelStick-5C695A1B5AE0-02.json)
](url)

[FileMgr] Configuration File: '/config.json' saved 703 bytes.
[FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
[FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
[FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
[FileMgr] Configuration File: '/input_config.json' saved 1436 bytes.
[FileMgr] Configuration File: '/input_config.json' saved 557 bytes.
[FileMgr] Configuration File: '/RestoredConfig.json' saved 23 bytes.
------ operator():ESPixelStick/src/WebMgr.cpp:334: Received RestoredConfig message ------
[FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
[FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
[FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
[FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
[FileMgr] Configuration File: '/output_config.json' saved 1436 bytes.
[FileMgr] Configuration File: '/output_config.json' saved 326 bytes.
[FileMgr] Configuration File: '/config.json' loaded.
[FileMgr] Configuration File: '/input_config.json' loaded.
[InputMgr] Shutting Down 'Disabled' on Input: 0
[InputMgr] Starting Artnet for channel '0'.
[Artnet] Subscribed to broadcast
[Artnet] Listening for 0 channels from Universe 1 to 0
[Artnet] Listening for 0 channels from Universe 1 to 0
[FileMgr] Configuration File: '/output_config.json' loaded.
[OutputMgr]  Shutting Down 'Disabled' on Output: 0
[OutputMgr] 'WS2811' Initialization for Output: 0
[Artnet] Listening for 396 channels from Universe 1 to 1
[   ESP] ***-----Internal Reboot Requested. Rebooting Now
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (520) esp_core_dump_flash: No core dum⸮⸮⸮⸮ѥѥ⸮⸮⸮found!
E (521) esp_core_dump_flash: No core dump partition found!

[   ESP] ESPixelStick v4.0-ci8327847865 (Mar 18 2024 - 13:59:28)
[   ESP] v4.4.6-dirty
[FileMgr] Flash file system initialized. Used = 225280 out of 327680
[FileMgr] Listing directory: /
[FileMgr] 'RestoredConfig.json': 	'23'
[FileMgr] 'UpdRecipe.json': 	'784'
[FileMgr] 'admininfo.json': 	'180'
[FileMgr] 'config.json': 	'703'
[FileMgr] 'input_config.json': 	'1993'
[FileMgr] 'output_config.json': 	'11814'
[FileMgr] Listing directory: //www
[FileMgr] 'alexa.html.gz': 	'146'
[FileMgr] 'apa102.html.gz': 	'1503'
[FileMgr] 'artnet.html.gz': 	'385'
[FileMgr] 'ddp.html.gz': 	'144'
[FileMgr] 'dmx.html.gz': 	'455'
[FileMgr] 'e1_31.html.gz': 	'460'
[FileMgr] 'effects.html.gz': 	'973'
[FileMgr] 'esps.css.gz': 	'13424'
[FileMgr] 'esps.js.gz': 	'62117'
[FileMgr] 'favicon.ico': 	'6450'
[FileMgr] 'fpp_remote.html.gz': 	'385'
[FileMgr] 'gece.html.gz': 	'501'
[FileMgr] 'gs8208.html.gz': 	'1579'
[FileMgr] 'index.html.gz': 	'5232'
[FileMgr] 'mqtt.html.gz': 	'712'
[FileMgr] 'relay.html.gz': 	'507'
[FileMgr] 'renard.html.gz': 	'528'
[FileMgr] 'serial.html.gz': 	'619'
[FileMgr] 'servo_pca9685.html.gz': 	'523'
[FileMgr] 'tls3001.html.gz': 	'1503'
[FileMgr] 'tm1814.html.gz': 	'1617'
[FileMgr] 'ucs1903.html.gz': 	'1580'
[FileMgr] 'ucs8903.html.gz': 	'1585'
[FileMgr] 'ws2801.html.gz': 	'1502'
[FileMgr] 'ws2811.html.gz': 	'1582'
[FileMgr] SD Card Size: 953MB
[FileMgr] Roma05.fseq - 12288064
[FileMgr] Configuration File: '/config.json' loaded.
[InputMgr] Merging Restored Input Config File
[FileMgr] Configuration File: '/input_config.json' saved 1609 bytes.
[FileMgr] Configuration File: '/input_config.json' loaded.
[OutputMgr] Merging Restored Output Config File
[FileMgr] Configuration File: '/output_config.json' saved 11796 bytes.
[FileMgr] Configuration File: '/output_config.json' loaded.
[FileMgr] Configuration File: '/admininfo.json' saved 180 bytes.
[FileMgr] Configuration File: '/config.json' saved 703 bytes.
[WiFiDrv] WiFi Entering State: Connecting As AP
[WiFiDrv] WiFi SOFTAP:       ssid: 'ESPixelStick-AP-esps-5C695A1B5AE0
[WiFiDrv] WiFi SOFTAP: IP Address: '0.0.0.0
[EthDrv] Entering State: Powering Up
[ 21327][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (21321) gpio: gpio_set_level(227): GPIO output gpio_num error
[EthDrv] Entering State: Connecting
[EthDrv] ETH IP Before Start: 0.0.0.0
[EthDrv] Ethernet Connecting as esps-5C695A1B5AE0
[EthDrv] Connecting to Ethernet using DHCP
[EthDrv] Entering State: Wait for IP Address
[EthDrv] Entering State: Got IP
[WebMgr] Web server listening on port 80
[  FPPD] FPPDiscovery subscribed to broadcast messages on port: 32320
[  FPPD] FPPDiscovery subscribed to multicast: 239.70.80.80:32320
[EthDrv] Ethernet Connected with IP: 192.168.72.137
[ 31646][E][vfs_api.cpp:105] open(): /littlefs/www/index.html does not exist, no permits for creation

MartinMueller2003, 2024-03-19:
Config merge has been disabled. It is a future looking feature and not needed at this 
time. Config gets restored now.

MartinMueller2003, 2024-03-19:
I am still working on a config merge mechanism that will work on significant upgrades. The current merge only works on the same version of SW that you started with. It will result on config fields not getting set when changed in the UI.