forkineye/ESPixelStick #752: esp32_devkitc GPIO settings

kingofmonkeys, 2024-03-12:
I've been trying to get the latest code working on an esp32 devkitc v4 and there appears to be a bug in how the GPIO pins are being defined in the GPIO_Defs.hpp file.  If you look at line 125 there appears to be a copy/paste mistake where its looking for BOARD_ESP32_TWILIGHTLORD_ETH instead of BOARD_ESP32_DEVKITC .  See snippet below. 

#elif defined (BOARD_ESP32_TWILIGHTLORD_ETH)
#   include "platformDefinitions/GPIO_Defs_ESP32_DevkitC.hpp"

noman1228, 2024-03-13:
I fixed this for that exact dev board. My pull request hasn't been merged yet but my fork will work for you in the meantime. I could also walk you through the lines/files that need to be edited to get you up and running.

kingofmonkeys, 2024-03-13:
I fixed it for myself, just wanted to get the bug report in so its fixed in the main branch for everyone.

MartinMueller2003, 2024-03-14:
Fixed in https://github.com/forkineye/ESPixelStick/pull/749

closed 2024-03-14 by MartinMueller2003