forkineye/ESPixelStick #754: Dig-Octa gets dynamic IP sometimes instead of static IP

marcobrianza, 2024-03-15:
**ESPixelStick Firmware Version**
4.0-ci8282977806

**Hardware Version**
Dig-Octa

**Binary release or compiled yourself?**
CI release

**Describe the bug**
The board is configured with this config
[ESPixelStick-5C695A1B5AE0-static-ip.json](https://github.com/forkineye/ESPixelStick/files/14613373/ESPixelStick-5C695A1B5AE0-static-ip.json)

The configuration has a static IP on the ethernet port and AP mode on Wi-Fi
Sometimes at board power up correctly applying a static ip with this log:

[FileMgr] Configuration File: '/config.json' saved 655 bytes.
[WiFiDrv] WiFi Entering State: Connecting As AP
[WiFiDrv] WiFi SOFTAP:       ssid: 'ESPixelStick-AP-esps-5C695A1B5AE0
[WiFiDrv] WiFi SOFTAP: IP Address: '0.0.0.0
[EthDrv] Entering State: Powering Up
[ 12551][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (12545) gpio: gpio_set_level(227): GPIO output gpio_num error
[EthDrv] Entering State: Connecting
[EthDrv] ETH IP Before Start: 0.0.0.0
[EthDrv] Ethernet Connecting as esps-5C695A1B5AE0
[EthDrv] Connecting to Ethernet with Static IP
[EthDrv] Entering State: Wait for IP Address
[EthDrv] Entering State: Got IP
[Artnet] Subscribed to broadcast
[Artnet] Listening for 400 channels from Universe 1 to 1
[WebMgr] Web server listening on port 80
[  FPPD] FPPDiscovery subscribed to broadcast messages on port: 32320
[  FPPD] FPPDiscovery subscribed to multicast: 239.70.80.80:32320
[EthDrv] Ethernet Connected with IP: 192.168.72.90

Sometimes instead the boars get an IP in DHCP with this log:

[FileMgr] Configuration File: '/config.json' saved 655 bytes.
[WiFiDrv] WiFi Entering State: Connecting As AP
[WiFiDrv] WiFi SOFTAP:       ssid: 'ESPixelStick-AP-esps-5C695A1B5AE0
[WiFiDrv] WiFi SOFTAP: IP Address: '0.0.0.0
[EthDrv] Entering State: Powering Up
[ 12856][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (12850) gpio: gpio_set_level(227): GPIO output gpio_num error
[EthDrv] Entering State: Connecting
[EthDrv] ETH IP Before Start: 0.0.0.0
[EthDrv] Ethernet Connecting as esps-5C695A1B5AE0
[ 16495][E][ETH.cpp:432] config(): STA IP could not be configured! Error: 20487
[EthDrv] Connecting to Ethernet with Static IP
[EthDrv] Entering State: Wait for IP Address
[EthDrv] Entering State: Got IP
[Artnet] Subscribed to broadcast
[Artnet] Listening for 400 channels from Universe 1 to 1
[WebMgr] Web server listening on port 80
[  FPPD] FPPDiscovery subscribed to broadcast messages on port: 32320
[  FPPD] FPPDiscovery subscribed to multicast: 239.70.80.80:32320
[EthDrv] Ethernet Connected with IP: 192.168.72.137

I cannot trace the cause of the issue. I have tried in powering the board with 24V since by USB it was causing issues sometimes, but the issue still presents itself.

marcobrianza, 2024-03-15:
Setting Power Active Delay (MS) to 2000 instead of 1000 looks like reduces issue but still happens sometimes, going to 3000 does not help further. I still think there should be a software level intervention to fix it

marcobrianza, 2024-03-15:
For the sake of tracking down the issue I also tested a WT32-ETH01 board and the issue is also present

MartinMueller2003, 2024-03-15:
The issue is happening in the lower level kernel code. Not sure there is much we can do about it at the application level.

This error comes from the kernel driver: 
[ 16495][E][ETH.cpp:432] config(): STA IP could not be configured! Error: 20487

I will take a look at what it is complaining about, but there will be very little I can do about it.

MartinMueller2003, 2024-03-15:
It looks like the issue should be fixed. However, based on your feedback it may still  be broken in the IDF version we are using. I added a few delays per https://github.com/espressif/arduino-esp32/issues/5733
You should be able to grab the build artifacts from my branch at 16:30 EDT and test.

marcobrianza, 2024-03-16:
Hi, I built your branch and did not get anymore a dynamic IP so far after at least 10 power cycles 🎉
great job as usual 🥇

marcobrianza, 2024-03-16:
While you are on the topic I noticed we don't have DNS entry for the static IP settings, I could turn handy in case someone needs for example to configure a MQTT server by name instead of IP

MartinMueller2003, 2024-03-16:
i will add them

marcobrianza, 2024-03-17:
I tested the latest CI release and consider this solved

MartinMueller2003, 2024-03-17:
Added support for manually configured Primary and secondary DNS addresses used while in Static IP mode. PR Created.

marcobrianza, 2024-03-18:
Hi I quickly tested the feature and came across a related issue in the interface where the Broker IP is limited and cannot use a long name

[image]

MartinMueller2003, 2024-03-18:
By design, we do not allow names in those fields they are intentionally limited to IP addresses.

marcobrianza, 2024-03-18:
do you intend to change it to allow names? if not the DNS filelds you just added becomes useless in my opinion.

MartinMueller2003, 2024-03-25:
I will add support for using names in the MQTT broker config.

marcobrianza, 2024-08-11:
I think this has been implmented

closed 2024-08-11 by marcobrianza