forkineye/ESPixelStick #756: universe zero in artnet mode doesn't seem to work

iz1kbp, 2024-03-29:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-ci8301555430

**Hardware Version**
d1_mini_pro

**Binary release or compiled yourself?**
Binary release 

*Operating System (and version)**
win10 N

**Web Browser (and version)**
firefox 124.0.1 /chrome 123.0.6312.86 (Build ufficiale) (a 64 bit

**Access Point**
direct to router

**Describe the bug**
Setting UNI ZERO in artnet mode to dmx, do not read any trasmitted data to out put dmx (set GPIO 17) .All ok on other uni.
In SACN modo uni 1 is ok.
Uni ZERO tested in parallel with another ArtNet interface

Mario

MartinMueller2003, 2024-07-09:
Fixed. Had code in there to specifically prevent the use of universe zero. This has been removed.

closed 2024-07-09 by MartinMueller2003

iz1kbp, 2024-07-10:
OK, Thank you again!