forkineye/ESPixelStick #757: Led for simple monitor  activity

iz1kbp, 2024-03-29:
**Is your feature request related to a problem? Please describe.**
No,it is not related to a problem

**Describe the solution you'd like**
A simple LED (light on) to monitor if it is connected to wifi and a led to identify if data in is present.

**Additional context**
If is possibile, may be a good and simple monitor (2 led on GPIO pin) during the installation without a pc o fixture connected: connect to wifi ok or not and data in present or not.
Thank again
Mario

MartinMueller2003, 2024-03-29:
Unfortunately, what you are asking for is not possible. The use of the LEds while in programming mode is completely out of our control. The boot loader built into the device from Espressive is in control of the CPU at that time. As for an LED on when receiving network traffic, those LEds become a problem in a show or environment where light pollution is a bad thing. I actually removed the LEDs from my buck converters because the power boxes were glowing. Black tape is on all of the LEDs on the controllers. This is why we have status on the UI to help debug network issues.

iz1kbp, 2024-03-29:
Ok,no problem!😉 Thank' anyway! Have a nice Easter!Mario

 

**********************************************************Mario Brustia - IZ1KBPLOC JN45IK - 45.4167N 8.6667E - QRZ Pilota SAPR ENAC A2 ITA-RP-000000609ABA / IT-STS-012008 OPERATORE ENAC:ITEAYYIFSKOPERATORE EASA:ITA3YNC64M41Y776R2SMPix4D certified user - QSC certified userP.IVA 02638890034 - Albo Artigiani N. ***@***.***@pec.itwww.meteotrecate.it  www.aprtrecate.itFacebook >> brustiamariosent from webmail**********************************************************

Data: Fri, 29 Mar 2024 11:05:29 -0700

closed 2024-03-29 by MartinMueller2003