forkineye/ESPixelStick #758: SACN problem in priority

iz1kbp, 2024-03-30:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-ci8301555430

**Hardware Version**
d1_mini_pro

**Binary release or compiled yourself?**
binary

**Operating System (and version)**
w10 and consol chamsys

**Web Browser (and version)**
FF and chrome (last)

**Access Point**
Lan 2.0.0.x in multiscast

**Describe the bug**
I wrote this text using my SHOWTEC NEt8/3 interface in sacn mode as a comparison. I inserted two consoles into the network that transmitted on the same universes but with different PRIORITIES: 100 (backup ,channel 1 at value 10) and 200 (primary ,channel 1 at value 255). On the showtec interface, when I transmit with the primary console, the DMX signal is at 255 and by acting on the secondary console the signal still remains at 255. When the primary console is disconnected and the secondary console enters, the dmx signal goes to 10 . using the ESP wifi system, with the two consoles connected in the same way, and both active, the DMX signal is unstable, the values ​​fluctuate between 255 and 10. If I disconnect the primary console, the signal goes to 10 , following the commands from the secondary console.It seems that the priority control has some problems
Mario

MartinMueller2003, 2024-03-30:
PixelStick does NOT support sACN priorities. It will send whatever data it receives and assumes only a single sACN source is sending data to it. This would be an enhancement request.

iz1kbp, 2024-03-30:
Ok! I operate in this way! Thank's for the support!

 

**********************************************************Mario Brustia - IZ1KBPLOC JN45IK - 45.4167N 8.6667E - QRZ Pilota SAPR ENAC A2 ITA-RP-000000609ABA / IT-STS-012008 OPERATORE ENAC:ITEAYYIFSKOPERATORE EASA:ITA3YNC64M41Y776R2SMPix4D certified user - QSC certified userP.IVA 02638890034 - Albo Artigiani N. ***@***.***@pec.itwww.meteotrecate.it  www.aprtrecate.itFacebook >> brustiamariosent from webmail**********************************************************

Data: Sat, 30 Mar 2024 08:35:12 -0700

MartinMueller2003, 2024-03-30:
This is just a note for a future implementer of this feature. It gives a nice description of how a product handles priority data. The one thing it does not mention is timing out a source. How long do you wait before moving to a lower priority data source.
https://support.etcconnect.com/ETC/Networking/General/Difference_between_sACN_per-address_and_per-port_priority

iz1kbp, 2024-03-30:
...on time out, in fact I have never read any data. Personally I use universe priority. I also tried channel priority but it is only implemented by some receivers and some "standard"received read this "gap" like a empty frame. Etc, in fact,  recommended not to use it in systems that do not implement the channel code.Priority  is certainly a useful option since it is essentially a much simpler backup than artnet. the intervention times? in my opinion two frames...like priority channel (1 frame)Have a nice Easter!Mario

 

**********************************************************Mario Brustia - IZ1KBPLOC JN45IK - 45.4167N 8.6667E - QRZ Pilota SAPR ENAC A2 ITA-RP-000000609ABA / IT-STS-012008 OPERATORE ENAC:ITEAYYIFSKOPERATORE EASA:ITA3YNC64M41Y776R2SMPix4D certified user - QSC certified userP.IVA 02638890034 - Albo Artigiani N. ***@***.***@pec.itwww.meteotrecate.it  www.aprtrecate.itFacebook >> brustiamariosent from webmail**********************************************************

Data: Sat, 30 Mar 2024 08:48:47 -0700

https://support.etcconnect.com/ETC/Networking/General/Difference_between_sACN_per-address_and_per-port_priority

MartinMueller2003, 2024-06-25:
Added some basic priority processing to the E1.31 module. It will not discriminate between two consoles at the same priority but it will require 5 missed frames from a higher level priority consol before accepting frames from a lower priority console.

Changes are in the "experimental" image on the web flash tool.

MartinMueller2003, 2024-07-09:
This has been merged.

closed 2024-07-09 by MartinMueller2003

iz1kbp, 2024-07-10:
Thank you Martin! ;)