forkineye/ESPixelStick #760: Issue running Output 4 with SD card on ESP32 Mini

jasonsomers, 2024-04-07:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-ci8499058366 (and several previous releases)

**Hardware Version**
ESP32 Mini on a RGB2Go Tetra2Go

**Binary release or compiled yourself?**
CI Release

**Operating System (and version)**
Windows 11  10.0.22631 Build 22631

**Web Browser (and version)**
123.0.6312.106 (Official Build) (64-bit)

**Access Point**

**Describe the bug**
I have been working on installing the firmware on an ESP D1 Mini32, trying to provide my users al alternative to WLED.
I keep running into the same issue on each build I try:
If I set up only the first 3 outputs. (GPIO 2,12,13) EVERYTHING works beautifully.
I can stream DDP with no issues.
I can upload FSEQs to the SD card using Xlight's FPP connect with no issues.
FPP Remote Functionality works great.

However, once I add Output 4, things start acting wonky.
FPP uploads stop working - getting 404 errors. Wireshark confirms the 404, and shows that the uploaded file is not found on the SD.
I can stream DDP ok.
Files pre-loaded on SD card are occasionally no longer visible
FPP remote functionality works for a second or two (assuming the file is already on the SD and still visible) and then it stops/crashes. I usually need to reboot ESPS after this.

I noticed that the default GPIO for Output 4 is GPIO5, which  conflicts with the SD card's CS pin, but didn't think anything of it since I immediately adjusted the GPIO to the one the Tetra2Go uses, GPIO 14.
 My SD card's pins are:
 CS: 5
 Clock: 18
 MISO: 19
 MOSI: 23
 
It seemed to me that despite the adjusted GPIO setting on Output4, that some signaling was still happening on GPIO5.
I hooked up a scope to GPIO5 and got the following with only the first 3 outputs configured: (Please excuse the scope not being zeroed properly)
https://youtu.be/TLkJ3aQ54JE
Signal looks to be turned high, with a few blips here and there

When I enabled Output 4,
https://youtu.be/wm5OZQ4vUSk
Enabling output 4 not only turns the signal low, but it also causes some data/garbage to be transmitted on GPIO5 that presumably throws off the CS pin of the SD card.

Please let me know if there are any other details I can provide to help you identify the issue.
-Jason

MartinMueller2003, 2024-04-07:
Please send a screenshot of the admin page. That will tell me exactly which platform build you are using and a snapshot of the device configuration page. 

Do you have a link to the HW platform that I can look at?

In the video, it looks like your scope is too slow and it is difficult to see the pattern coming out. Can you capture the burst of transitions? FYI: I have given up using a scope and use a  logic analyzer. 
If the transitions are correct, then there should be a stream of transitions with a 1.2us period. Most 1Mhz scopes cannot properly see these transitions.

jasonsomers, 2024-04-07:
Yeah - I totally understand. Without a signal analyzer available, I was just using what I had on hand to try and explain the behavior I was seeing.  I guess the part that struck me was that the signal was DIFFERENT when Output4 was enabled, and it shouldn't be. The rest is just rationalization and strategic guessing.

The hardware platform is my controller, the Tetra2Go. It is a simple 4-port ESP32-based pixel controller.
https://rgb2go.com/products/tetra2go-4-port-wi-fi-pixel-controller
Outputs are configured for GPIOs: 2,13,12,14
SD Card is on a separate module that is not yet publicly available. Pinouts are in my original post.

I realize that my controller is an unknown here, so I will try and reproduce the issue without my board in the middle. If I am correct. we should see the same behavior while connecting directly to a ESP32 Mini with a standard SPI SD Card reader module. I will work on that and update you.

[image]
[image]
[image]

jasonsomers, 2024-04-07:
FYI - I have reproduced the same behavior without my controller.
I have wired the SD card module to the ESP32 via a breadboard, and am seeing the same issue.
3 outputs work fine (judging by the "Diagnostic" output).
Once I add the 4th output, SD card operations start becoming compromised.

Hopefully this should be easy enough to reproduce.
[image]

MartinMueller2003, 2024-04-07:
I am running a dig quad and a dig octa without issue. That is why I need to see the Admin page so I can figure out the interactions you are seeing.

MartinMueller2003, 2024-04-08:
I am creating a platform profile for your HW. It will be in the flashtool as a target platform.

MartinMueller2003, 2024-04-08:
I just uploaded your platform to my git repository. It is not yet integrated into the flash tool but you should be able to build and load the image. It is ESP32_tetra2go.

MartinMueller2003, 2024-04-08:
Finished adding your platform to the build. It is now part of the dist found at:
https://github.com/MartinMueller2003/ESPixelStick/releases/tag/dist

I suspect you will stop seeing GPIO conflicts. Let me know what you find.

jasonsomers, 2024-04-09:
Everything working out of the gate on my first attempt.
Excellent!

Thanks so much for your time an efforts!

MartinMueller2003, 2024-04-09:
The simple answer is that you missed some GPIO settings that conflicted with what you were doing.

jasonsomers, 2024-04-09:
Can you be more specific? Which settings could I have missed?
I confirmed that all output GPIOs were set properly in the GUI, and the SD
pins were also set properly.
I had two different users confirm this.

Were these settings not available in the GUI? If so that would make sense.
I did not try to write my own platform files and recompile.

Just curious - always looking to learn.
Thanks again for your help.
Jason

MartinMueller2003, 2024-04-09:
I suspect it was in the SPI configuration for the WS2800 protocol. That or one of the I2C ports. I have found over time that it is best to make a new platform where you can see all of the settings on one page and verify there are no overlaps. I did not dig deeply into your config, I just took the steps that I know work for me.