forkineye/ESPixelStick #761: GPIO problems with wemos D1 mini and Pro

iz1kbp, 2024-04-19:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-ci8301555430

**Hardware Version**
d1_mini_pro and d1_mini

**Binary release or compiled yourself?**
Mar 15 2024 - 19:48:24

**Operating System (and version)**
w10n 64

**Web Browser (and version)**
FF

**Access Point**

**Describe the bug**
Hi Manuel, I'm doing some tests with wemos d1 mini and wemos d1 mini pro (external antenna).

By setting out gpio 16 on d1 mini at high level (for a led on -> enable output relay> enable GPIO 16 >sigend Inverter> threshold 255 >> have the led on), the setting upon reboot, while on wemos d1 pro, when I set the output the led was on, but on reboot ,even if the panel appears correct, the output level is at low level (led off).
Simply I need to set a gpio (d0 or d1 or d2) to a high value  as MONITOR to understand that the interface is on and working. Maybe I'm doing something wrong?
thank's in adavance
Mario

iz1kbp, 2024-04-19:
... basically. I use the same settings on WEMOS d1 mini and Pro. upon reboot, while the MINI maintains the GPIO outputs as programmed (ON), the PRO does not maintain them (OFF)

MartinMueller2003, 2024-06-25:
Not sure what the issue is here. By default the GPIOs on an ESP are "Inputs" after a reset. That means the state high/low is indeterminate (high impedance) and can drift high or low as the outside circuit wants. If you need a GPIO to be in a specific state after a reboot and before the SW initializes the ports then you need to add a pullup/pulldown resistor to keep the GPIO at the desired state until the SW is in control.

MartinMueller2003, 2024-06-25:
FYI, if you pull GPIO0 or GPIO 2 in the wrong direction, your device will not boot.