forkineye/ESPixelStick #765: Flashing a Wemos D1 mini and setting up a relay  GPIOD set to 4294967295  by default

patdelaney, 2024-06-22:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-ci9115820340  Wemos D1 MINI

**Hardware Version**
Wemos D1 MINI

**Binary release or compiled yourself?**
Web Flasher

**Web Browser (and version)**
Chrome

**Describe the bug**
I went to update a Wemos D1 Mini that has a singleport relay hat attached. After flashing with the web tool, I went in to configure it for a relay. The Default screen after you select relay for Output 2 shows errored out numbers.  4294967295  by default

I'm guessing that these should actually be 0
[image]

Also in the Webflasher tool I configured ESPRelay as the Device Name  but after it flashed it reverted to default ESPixelstick

MartinMueller2003, 2024-06-22:
That number represents -1. You need to give valid numbers to move on.

patdelaney, 2024-06-22:
So populating those red boxes with that number after flashing is sxpected behavior? I would think they shoud be empty and only force a value if enabled is checked.

MartinMueller2003, 2024-06-25:
The config was being saved as 4Billion as opposed to -1. This is fixed in todays experimental version on the web flasher and will be in an upcoming CI build.

closed 2024-06-25 by MartinMueller2003