forkineye/ESPixelStick #766: Default Ethernet configuration for Wasatch board with Ethernet is incorrect.

patdelaney, 2024-06-23:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-ci9475001076

**Hardware Version**
Wasatch Quad+ 4 onboard ports and  4 Differential Ports

**Binary release or compiled yourself?**
Used Web flasher

**Describe the bug**
When I flashed the firmware (4.0-ci9475001076) I did configure Wi-Fi. After it rebooted and I verified that it would connect. I put the ESP back on the board and connected the ethernet cable, then powered it up. It appears to have reconnected via Wi-Fi and not ETH0.
 
I research the board and the correct settings are below and I tested them with that board.
[image]

patdelaney, 2024-06-24:
[image]

MartinMueller2003, 2024-06-24:
Corrections for this will be in the next PR.

MartinMueller2003, 2024-07-09:
This has been merged

closed 2024-07-09 by MartinMueller2003