forkineye/ESPixelStick #767: Default Ethernet configuration for LED output GPIO pins Wasatch Quad+ board  (port 8) is incorrect.

patdelaney, 2024-06-24:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-ci9475001076

**Hardware Version**
Wasatch Quad+ 4 onboard ports and 4 Differential Ports

**Binary release or compiled yourself?**
Used web flasher

**Describe the bug**
The version I installed has GPIO3 listed for Output 8. The documentation lists 33.

I changed the GPIO to 33 and lights work as expect.

patdelaney, 2024-06-24:
[image]

MartinMueller2003, 2024-06-24:
A fix for this is available on the web flasher (experimental" version and will be part of the next pull request.

MartinMueller2003, 2024-07-09:
This has been merged

closed 2024-07-09 by MartinMueller2003