forkineye/ESPixelStick #774: Using WEB APP >> set as static IP but the programmed IP is different (my first DHCP address in this case)

iz1kbp, 2024-07-11:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
Using the WEB APP and lastest release

**Hardware Version**
WEMOS D1 MINI PRO 16M

**Binary release or compiled yourself?**
DOWNLOAD FROM WEBAPP

**Operating System (and version)**
W11

**Web Browser (and version)**
CHROME LASTEST

**Access Point**
VIA LAN

**Describe the bug**
I tried programming via WEB using last release available. The programming was successful, but although I had programmed a static IP, in the programming log I saw that an IP was set which, in my case, corresponds to the first of the DHCP range
My setup program..
[image]
log file (2.0.0.100 insted of 2.0.0.28)
[image]

the release
[image]

MartinMueller2003, 2024-07-12:
Found two major issues. Both are fixed.

1. Data was being sent from the UI to the server incorrectly formated/ This resulted in all of the bool settings being corrupted and reverting to factory defaults. 
2. Validation functions could get confused and leave the flash button disabled if a setting is given an invalid value and then cleaned up later.

closed 2024-07-12 by MartinMueller2003

iz1kbp, 2024-07-13:
thank you!
:+1: