forkineye/ESPixelStick #776: Home screen shows no data on v4-beta5

AGFazio, 2024-07-19:
**ESPixelStick Firmware Version**
ESPixelStick Firmware v4 - beta5

**Hardware Version**
espsv3

**Binary release or compiled yourself?**
Bianry

**Operating System (and version)**
Windows 11

**Web Browser (and version)**
Edge Version 126.0.2592.113 (Official build) (64-bit)

**Access Point**

**Describe the bug**
When on web home page, pixelstick shows no data. It loads for a short time then when refresh the page it dissapears. 
[image]

MartinMueller2003, 2024-07-20:
Please load the most recent CI build from the Web Flasher. That is a bug that has been fixed.

AGFazio, 2024-07-20:
Done.  Works.  Thank You.

closed 2024-07-23 by MartinMueller2003