forkineye/ESPixelStick #778: ftp connection triggers reboot

marcobrianza, 2024-07-28:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**

4.0-ci10023791775 
Jul 20 2024 - 23:21:23

**Hardware Version**
D32 PRO

**Binary release or compiled yourself?**
web installer

**Describe the bug**

hello, I am catching up with the latest features and started trying ftp using Cyberduck
as soon as i try the connection the board reboots with the following log

...

assert failed: xQueueGenericSend queue.c:832 (pxQueue->pcHead != ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == xTaskGetCurrentTaskHandle())

Backtrace: 0x400846ed:0x3ffe0f70 0x4008ea05:0x3ffe0f90 0x40094d19:0x3ffe0fb0 0x4008f4ba:0x3ffe10e0 0x401178d9:0x3ffe1120 0x4010ccd1:0x3ffe1140 0x40111335:0x3ffe1160 0x40110b69:0x3ffe1180 0x4011110a:0x3ffe11a0 0x40110d23:0x3ffe11c0 0x40110d39:0x3ffe11e0 0x401110ac:0x3ffe1210 0x40111208:0x3ffe1230 0x4011125a:0x3ffe1250 0x40111433:0x3ffe1270 0x4010ea49:0x3ffe1290 0x400d2959:0x3ffe12c0 0x40115484:0x3ffe12f0 0x401154ad:0x3ffe1310 0x400d3185:0x3ffe1330 0x400d3315:0x3ffe13b0 0x400e6679:0x3ffe1430 0x4008051e:0x3ffe1530 0x400e5de9:0x3ffe1580 0x400e5e07:0x3ffe15a0

ELF file SHA256: d62b1cb3cf54435f

E (13965) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (461) esp_core_dump_flash: No core dum���ѥѥ���found!
E (462) esp_core_dump_flash: No core dump partition found!

[   ESP] ESPixelStick v4.0-ci10023791775 (Jul 20 2024 - 23:21:23)
[   ESP] v4.4.7-dirty

MartinMueller2003, 2024-07-28:
Fix will be in the next PR. There was a missing flag in the list of fseq files going to the filemanager screen. The flag was present in the status messages on the home page. Going to the device page cleared the flag and turned off the file admin page.

MartinMueller2003, 2024-07-28:
Sorry. This did not fix the FTP issue.

MartinMueller2003, 2024-07-28:
Just tested with my latest CI build (built today) and it is working fine on my Lolin D32 Pro with a 16GB Sandisk Micro SD

MartinMueller2003, 2024-07-28:
Using Cyberduck

[image]

marcobrianza, 2024-07-28:
the last CI release I see is this:

FW Version4.0-ci10058255834Build 
DateJul 23 2024 - 11:44:59
ArchitectureESP32
Board Named32_pro

with this version the boards does not reboot but the ftp service is not fully working. I you there is a newer version I would go testing that one.

side note: some issues may be related to the sd card my od 2GB card has more issues compared to a newer sandisk 16GB
I will test them in future releases

MartinMueller2003, 2024-07-28:
You should be testing with the latest CI build on the Web Flash tool. That would be dated July 28 2024

marcobrianza, 2024-07-28:
I did not see the July 28 at first because it was not sequential by date in the selector.
using 
FW Version4.0-ci10132700281 
Build Date Jul 28 2024 - 15:08:14

connection, file listing, and delete works for me, but not upload and download sequences

MartinMueller2003, 2024-07-28:
That is the version I tested and uploaded just fine

marcobrianza, 2024-07-29:
Insisted in trying using another ftp client (simple client for mac) and fond this working flawlessly

marcobrianza, 2024-08-07:
I have noticed that the connection issue may be related to ftp clients opening multiple connections at the same time while esp supports only one connection. Does this look resonable for you?

MartinMueller2003, 2024-08-07:
I always had to manually configure the client to only use a single connection. Anything else cause odd issues.

marcobrianza, 2024-08-11:
I'm closing this issue, but I think that the single connection limitation should be mentioned in the documentation

closed 2024-08-11 by marcobrianza