forkineye/ESPixelStick #780: Evo Pixels not working with ESPixelStick

joe8012, 2024-08-12:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
ci-10229501118

**Hardware Version**
Dig - Octa
Dig - Quad AE+

**Binary release or compiled yourself?**
No - Web Installer

**Operating System (and version)**
Windows 11 v 10.0.22631

**Web Browser (and version)**
Chrome

**Access Point**
Wifi

**Describe the bug**
I have ESPixelStick version ci-10229501118 installed on multiple dig-octa and dig-quad ae+ controllers.  Everything worked as expected until I changed some of my 12v 12mm regulated bullet pixels to 12v evo’s.  The evo’s would not light and when changing the resistor switch on the controller, they would sometimes light but dimly flashing on and off randomly with various colors.
The evo’s worked as expected on all these controllers when the WLED firmware is flashed to them.  I only have this problem when ESPixelStick is flashed.  In further testing, the evo’s do work with the ESPixelStick firmware when a sacrificial 12mm bullet pixel is placed between the controller and the evo’s.  However, when I placed a dig 3-wire data booster in line with the evo’s, they did not light?  I prefer not to use a sacrificial pixel as a work around on my 2400 pixel mega tree.  
I tested driving the pixels from the ESPixelStick UI, from the FPP player and directly from Xlights with the same results.  Thank You for any help with the issue.

patdelaney, 2024-08-13:
I'm running EVOs on a ForkinEye pixelstick. I have some digquad boards at home, I can test on those when I get home from work. How many outputs are you using on the DIG boad?

MartinMueller2003, 2024-08-13:
> I'm running EVOs on a ForkinEye pixelstick. I have some digquad boards at home, I can test on those when I get home from work. How many outputs are you using on the DIG boad?

Which processing board are you using?

joe8012, 2024-08-13:
All 8 on a dig octa, 300 evo's per channel 2400 total.  You get the same result with just one strand plugged into one port.  Also same result with one strand plugged into a dig quad ae+.

patdelaney, 2024-08-13:
I just got home, and for grins I tested esp32_quinled_quad_ae_plus_8 running 4.0-ci9732852095 Jun 30 2024 - 14:51:12. I'm not getting any output. ANd when I send e1.31 data. I have regular bullets connected and I get no output. I'm sending data and nothing lights.  I tried to run an effect thta is not working.  I'll upgrade to a newer version that is availabke in the web updater

joe8012, 2024-08-13:
I have only tried ddp.

patdelaney, 2024-08-13:
I'm testing with the AUG 6 build and I can get evo output, but it's glitchy. I can send E1.31 data or I can run rainbow and there is something odd going on with random flashes.  this device is wi-fi.
https://drive.google.com/file/d/1coZAoaoUExU-UF_CJ008Xx7rpKl5001Z/view?usp=drive_link
https://drive.google.com/file/d/1DDOQU57QG1MFe2qAqUOsB_KCwuVfwUlZ/view?usp=drive_link

See these example videos

I will test on other hardware

patdelaney, 2024-08-13:
4.0-ci9475001076  June 12 on RGB2go SEEMS TO WORK OK not glitch with bullets,  that unit is currently set up for 5 v   I need to flip it to 12v to test

joe8012, 2024-08-13:
Try your bullet pixels in line before the evo's.

joe8012, 2024-08-16:
I thought I would add an additional testing note.  While running the Espixelstick firmware, I tried to use a "resistor" style 12mm bullet sacrificial pixel between the controller and the evo's and it did not work as well.  I got random flickering and it got worse as you increased the brightness beyond 30%.   When I used the "regulated" 12mm bullet pixel it worked as expected all the way to 100% brightness.

joe8012, 2024-08-16:
It is now working with the Aug 16 CI firmware.  Thank you so much!

MartinMueller2003, 2024-08-16:
I changed the timing a bit to align with the sloppy timing generated by WLED. I will submit changes to main.

closed 2024-08-16 by MartinMueller2003