forkineye/ESPixelStick #784: Clicking Advanced Mode Has No Effect

gothambeat, 2024-09-11:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-beta3

**Hardware Version**
ESPIXELSTICK V3

**Binary release or compiled yourself?**
Binary Release

**Operating System (and version)**
Windows 11

**Web Browser (and version)**
Google Chrome

**Access Point**
N/A for this issue but UBiquiti managed VLAN network

**Describe the bug**
When selecting the "Advanced Mode" on the administration page,  nothing happens.

MartinMueller2003, 2024-09-11:
please retry using the latest ci build. i used this yesterday without issue

MartinMueller2003, 2024-09-12:
Just tested this again and it is working fine. NOTE: When you select advanced mode on the admin page, nothing visible will happen. When you navigate to the device page, you will now see the GPIOs used for the SD card.

MartinMueller2003, 2024-12-07:
It still works fine for me.

closed 2024-12-07 by MartinMueller2003