forkineye/ESPixelStick #788: wt32 01 serious disconnection, high RJ45 connection delay

517002650, 2024-10-10:
Has anyone encountered this problem? Or is it my personal problem? When using RJ45 to connect and ping IP, the packet loss delay is very large

MartinMueller2003, 2024-12-07:
I am getting ready to remove the WT32 from the supported platform list. It is supper unstable. At this time I am not fixing issues related to the WT32.

dutch-anykey, 2024-12-08:
Weirdly enough I have now a wt32 running on my testbench. 4 strings attached and running a test sequence from xlights via DDP. And a dos box open with a continuous ping. But no timeouts or delays on that ping via ethernet.

FW Version 4.0-ci12200963303Build 
Date Dec 6 2024 - 14:59:58
Architecture ESP32Board 
Name esp32_wasatch

Maybe try to disable wifi? Not sure if some weird internal routing issue could occur between the two interfaces.

dutch-anykey, 2024-12-19:
> You are one of the fortunate wt32 users. Most have nothing but complaints. I would toss mine in the trash if I did not need it for testing.

Oh I had a lot of trouble as well, but that's when trying to enable SD support. Haven't had any stable situation with that. 
Without SD card it seems stable.

MartinMueller2003, 2024-12-28:
Not going to spend any time on the WT32 platform.

closed 2024-12-28 by MartinMueller2003