forkineye/ESPixelStick #794: ESPixelstick 4.0-ci trying to get LAN + SDcard + 4 ws2811 outputs working on an WT32-eth01 board but having GPIO issues

dutch-anykey, 2024-11-18:
**ESPixelStick Firmware Version**
4.0-ci11448233108

**Hardware Version**
esp32_wt32eth01

**Binary release or compiled yourself?**
Binary release

**Operating System (and version)**
Windows 11 Pro

**Web Browser (and version)**
Version 131.0.6778.69

**Access Point**
TP-link omada platform

**Describe the bug**

I'm not able to get all the features working on a wt32:

- LAN
- 4 x ws2811 outputs
- SD card

Somehow I can get each working separately but not all features together due to GPIO assignments problems.
For now I have LAN + 4 x ws2811 working. But when I want to add SD support It won't boot, or won't start LAN or any other combination.

Question is if there are things that I have to keep in mind when assigning GPIO's?
on the board the following GPIO's are physically available:
0, 39, 36,15,14,12, 35, 4, 2

for LAN interface the GPIO's (non external GPIO's I guess) 17, 23, 18, 16 and 1 are assigned. I didn't change anything there.

For SD card a few predefined GPIO's were assigned, but I can't remember the exact numbers. I think mine were:
SD MISO Pin 12
SD MOSI Pin 13
SD Clock Pin 14
SD Chip Select Pin 15
For MOSI I changed to another free physical port since GPIO 13 is not on the boards output on the WT32

For led outputs:
ws2811 port1  GPIO 2
ws2811 port1  GPIO 4
ws2811 port1  GPIO 36
ws2811 port1  GPIO 39

It all was not working. I had various outcomes when changing GPIO numbers around for led and SD card connections. Mostly that resulted in LAN not being initialized and the other result was the board not booting. (I wanted to include the serial console output,but I  forgot to save it. If needed I can reproduce and include it)

Next step I did was skipping the need for SD card. So I configured non existing GPIO numbers:
SD MISO Pin 8
SD MOSI Pin 10
SD Clock Pin 9
SD Chip Select Pin 11

for WS2811 outputs:
ws2811 port1  GPIO 2
ws2811 port1  GPIO 4
ws2811 port1  GPIO 14
ws2811 port1  GPIO 15

Now all seems to be okay. (I haven't checked yet if those led GPIO's are realy outputting data)

So question is, should it be possible to have all functions working (SD, LAN, 4 WS2811 outputs)? And to even stretch it more have the last free GPIO available for  en external button.?

kind regards,
Maurice

dutch-anykey, 2024-11-18:
Hi Martin

So I learned something about GPIO's and the WT32 board.
What I want cannot be done because of the limited GPIO's on the wt32 board.

I can not have 4 ws2811 outputs + LAN +SD. Simply because there are not enough physical ports on the WT32. Also GPIO 0 I can not use for ws2811 output. Because somehow the LAN is working (showing up in the gui), but I get no IP via DHCP (weird right?)

So best I can do is LAN + SD and two ws2811 outputs:

GPIO's available on WT32-eth0
0    WS2811 output (but LAN get's no IP address somehow)
39  not used
36  not used 
15  SCLK
14  CS
12  MOSI
35  MISO
4    WS2811 output
2    WS2811 output

For my project here I will skip the SD card implementation. So I will send realtime DDP traffic over the ethernet port. Plan was to use it in multisync mode and play the sequences locally from the SD card. But I think the LAN connection will be fast enough to handle live ddp data.

Thanks for the info and the quick reply Martin. Besides the strange thing with GPIO 0 my questions are answered.

kind regards,
Maurice

MartinMueller2003, 2024-11-18:
Did you look at the WT32 implementation that is already part of the build set? All you would have to do is customize the GPIO Defs file for the WT32 board type. Yes you CAN use GPIO zero as a pixel output. You need to add a 4K pull up from GPIO zero to the 3.3v pin on the WT32 board so that the device will boot properly. Same is true for GPIO two.

dutch-anykey, 2024-11-18:
Hi Martin,

The more I google, the more I get to know. The WT32 board has more IO ports. It just has other names printed on the PCB instead of just "IO+number".
Lots more to fiddle :)
Sorry for my beginners questions. I will have a look at the build sets.

kind regards,
Maurice

MartinMueller2003, 2024-11-18:
If you look at the pre built version, you will see that it has 5 output ports defined. And supports 16 external PWM channels AND supports the SD card and has the Ethernet port working.

//Output Manager
#define DEFAULT_RMT_0_GPIO     gpio_num_t::GPIO_NUM_2
#define DEFAULT_RMT_1_GPIO     gpio_num_t::GPIO_NUM_4

#define DEFAULT_RMT_2_GPIO      gpio_num_t::GPIO_NUM_0
#define DEFAULT_RMT_3_GPIO      gpio_num_t::GPIO_NUM_5
#define DEFAULT_RMT_4_GPIO      gpio_num_t::GPIO_NUM_1

// #define SUPPORT_SPI_OUTPUT
#define DEFAULT_I2C_SDA         gpio_num_t::GPIO_NUM_3
#define DEFAULT_I2C_SCL         gpio_num_t::GPIO_NUM_17

// File Manager
// #define SUPPORT_SD
#define SD_CARD_MISO_PIN        gpio_num_t::GPIO_NUM_12
#define SD_CARD_MOSI_PIN        gpio_num_t::GPIO_NUM_13
#define SD_CARD_CLK_PIN         gpio_num_t::GPIO_NUM_14
#define SD_CARD_CS_PIN          gpio_num_t::GPIO_NUM_15

dutch-anykey, 2024-11-18:
Hi Martin,

Thanks for the help so far. It's all starting to make more sense.
I took a look at the pre built version and found all the IO's on the board except for GPIO13. I have read several documents about the WT32. But GPIO13 is a mystery to me. I'm missing something obviously. 

kind regards,
Maurice

dutch-anykey, 2024-11-20:
Hey Martin,

A lot has changed in two days, you really got me haha.
bought a breadboard so I don't have to solder wires for each change. Installed visual studio code and got it going eventually....
I'm 51 years old and for the first time in my life compiled some software :)
Still learning and looking around everywhere, stuff gets confusing really fast.

But so far I'm still not able to get the SD card going.
So far I'm going with the info you gave me so far:

- LAN GPIOS. These are fixed. You have no choice. The base hardware defines these and you cannot do anything else. The 
exception is the power select pin. That can be implementation dependent.
- SD CArd GPIOs. Lots of flexibility as long as you do not use GPIO 0 or 2
- GPIOs > 33 are input only. MISO from SD card can be on one of these

And other info in the posts. So in the WT32 build set file GPIO 13 is used as default. Is that tested on a real board? Because GPIO13 is really not available on the board as an external pin as far as I can seen on all WT32 docs I found.
But anyway I compiled some version where I took other GPIO pins. But no bueno so far. 
The other thing is the pull up resistors for various pins. You mentioned GPIO0 and GPIO2 they need to have a certain state during boot as I understand. But this resistor is okay also during "normal" operation? Or should there be some kind of mechanism to have it only connected during boot (I understand that it is also depending on what is really connected to the pin in/output)

Sorry for the questions. I can understand if you say that this is not a learning platform here and only for bug reports. If so than please close this (Or should I do it as I'm the one that created it?)

kind regards,
Maurice

:)

dutch-anykey, 2024-11-23:
Hi,

To anybody :)

This WT32 board is giving me headaches hehe

I'm reading up on compiling espixelstick. I thought I had it going, but from here https://github.com/forkineye/ESPixelStick . Regarding the filesystem, is the part about "the Arduino IDE should be used for uploading" still valid?

So far no success with getting an SD card working. I check with a "normal" ESP32 and the same SDcard reader and sd card is working. 
disabled a lot of other inputs in the Defs file for the WT32 and tried various pin assignements for SD:
#define SUPPORT_ETHERNET

//Output Manager
#define DEFAULT_RMT_0_GPIO     gpio_num_t::GPIO_NUM_2
//#define DEFAULT_RMT_1_GPIO     gpio_num_t::GPIO_NUM_4

// #define DEFAULT_RMT_2_GPIO      gpio_num_t::GPIO_NUM_0
// #define DEFAULT_RMT_3_GPIO      gpio_num_t::GPIO_NUM_12
// #define DEFAULT_RMT_4_GPIO      gpio_num_t::GPIO_NUM_1

// #define SUPPORT_SPI_OUTPUT
// #define DEFAULT_I2C_SDA         gpio_num_t::GPIO_NUM_3
// #define DEFAULT_I2C_SCL         gpio_num_t::GPIO_NUM_17

// File Manager
// #define SUPPORT_SD
#define SD_CARD_MISO_PIN        gpio_num_t::GPIO_NUM_39
#define SD_CARD_MOSI_PIN        gpio_num_t::GPIO_NUM_32
#define SD_CARD_CLK_PIN         gpio_num_t::GPIO_NUM_4
#define SD_CARD_CS_PIN          gpio_num_t::GPIO_NUM_33

#include <ETH.h>

/*
   * ETH_CLOCK_GPIO0_IN   - default: external clock from crystal oscillator
   * ETH_CLOCK_GPIO0_OUT  - 50MHz clock from internal APLL output on GPIO0 - possibly an inverter is needed for LAN8720
   * ETH_CLOCK_GPIO16_OUT - 50MHz clock from internal APLL output on GPIO16 - possibly an inverter is needed for LAN8720
   * ETH_CLOCK_GPIO17_OUT - 50MHz clock from internal APLL inverted output on GPIO17 - tested with LAN8720
*/
#define DEFAULT_ETH_CLK_MODE           eth_clock_mode_t::ETH_CLOCK_GPIO17_OUT

// Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
#define DEFAULT_ETH_POWER_PIN          gpio_num_t::GPIO_NUM_16
#define ETH_POWER_PIN                  DEFAULT_ETH_POWER_PIN
#define DEFAULT_ETH_POWER_PIN_ACTIVE   HIGH

// Type of the Ethernet PHY (LAN8720 or TLK110)
#define DEFAULT_ETH_TYPE               eth_phy_type_t::ETH_PHY_LAN8720

// I2C-address of Ethernet PHY (0 or 1 for LAN8720, 31 for TLK110)
#define ETH_ADDR_PHY_LAN8720           1

#define DEFAULT_ETH_ADDR               ETH_ADDR_PHY_LAN8720
#define DEFAULT_ETH_TXEN               gpio_num_t::GPIO_NUM_21
#define DEFAULT_ETH_TXD0               gpio_num_t::GPIO_NUM_19
#define DEFAULT_ETH_TXD1               gpio_num_t::GPIO_NUM_22
#define DEFAULT_ETH_CRSDV              gpio_num_t::GPIO_NUM_27
#define DEFAULT_ETH_RXD0               gpio_num_t::GPIO_NUM_25
#define DEFAULT_ETH_RXD1               gpio_num_t::GPIO_NUM_26
#define DEFAULT_ETH_MDC_PIN            gpio_num_t::GPIO_NUM_23
#define DEFAULT_ETH_MDIO_PIN           gpio_num_t::GPIO_NUM_18

// Output Types
// Not Finished - #define SUPPORT_OutputType_TLS3001
// #define SUPPORT_OutputType_APA102         // SPI
#define SUPPORT_OutputType_DMX            // UART
#define SUPPORT_OutputType_GECE           // UART
#define SUPPORT_OutputType_GS8208         // UART / RMT
#define SUPPORT_OutputType_Renard         // UART
#define SUPPORT_OutputType_Serial         // UART
#define SUPPORT_OutputType_TM1814         // UART / RMT
#define SUPPORT_OutputType_UCS1903        // UART / RMT
#define SUPPORT_OutputType_UCS8903        // UART / RMT
// #define SUPPORT_OutputType_WS2801         // SPI
#define SUPPORT_OutputType_WS2811         // UART / RMT
#define SUPPORT_OutputType_Relay          // GPIO
#define SUPPORT_OutputType_Servo_PCA9685  // I2C (default pins)

But no success

yesterday I tried to flash ESP4.0 beta 5 and that one won't even start-up
Maybe someone can confirm that he/she had an SD-card working on a WT32-eth0? 

br 
Maurice

dutch-anykey, 2024-11-26:
Please don't tell me I forgot to uncomment this line ??:
// #define SUPPORT_SD

pfffffffffffffff

Another thing. I'm not able to compile anymore. I get a lot of these kind of errors:

ESPixelStick/src/network/WiFiDriver.cpp:316:27: error: assignment of read-only location '*(((const char*)(& CN_passphrase)) + ((sizetype)(& json)->ArduinoJson::V721PB22::JsonObject::operator bool()))'

So I can't try to compile and upload my new config.

MartinMueller2003, 2024-11-26:
Yes you need to uncomment that for SD support.

That is an odd error message. I would manually delete the entire .pio directory.

1. Turn off VS Code
2. delete .pio
3. Turn on VS code
4. Wait for it to finish updating some of the files you deleted
5. Try a new build.

dutch-anykey, 2024-11-26:
Hi Martin,

That didn't bring a solution. I also tried to open it again in a new folder with a fresh zipfile of espixelstick. Let everything install, but still the same problem.
Tomorrow I will start fresh and reinstall VScode.
Also, is the install of nodejs and Gulp still applicable as described in the build requirements  (platform IO instructions on the web page)

thanks,
Maurice

forkineye, 2024-11-26:
> Another thing. I'm not able to compile anymore. I get a lot of these kind of errors:
>
> ESPixelStick/src/network/WiFiDriver.cpp:316:27: error: assignment of read-only location '_(((const char_)(& [...]

It's broke in the pipeline as well and for me locally. Something with the ArduinoJson updates, I'm looking into it.

forkineye, 2024-11-27:
There was an error in how lib_deps had git repo versions pinned. It was grabbing ArduinoJson 7.2.1 which breaks. I just pushed a fix to pin it to 7.2.0 for now - https://github.com/bblanchon/ArduinoJson/blob/7.x/CHANGELOG.md#breaking-changes

dutch-anykey, 2024-11-27:
Thanks,

I can build again :) It's working.
Now I can get on getting this SD working on the wt32 board

dutch-anykey, 2024-11-27:
Hi,

Still no luck with enabling the SD card. 
After my blunder with "// #define SUPPORT_SD"
I now at least get an error in the console:

[FileMgr] No SD card installed
[FileMgr] SD initialization failed - code: 1
Card reset failed

Same adapter and card works on a "normal" ESP32, but on my WT32 no luck. Tried different GPIO's. Currently:
#define SUPPORT_ETHERNET

//Output Manager
#define DEFAULT_RMT_0_GPIO     gpio_num_t::GPIO_NUM_2
#define DEFAULT_RMT_1_GPIO     gpio_num_t::GPIO_NUM_12

#define DEFAULT_RMT_2_GPIO      gpio_num_t::GPIO_NUM_32
#define DEFAULT_RMT_3_GPIO      gpio_num_t::GPIO_NUM_33
// #define DEFAULT_RMT_4_GPIO      gpio_num_t::GPIO_NUM_1

// #define SUPPORT_SPI_OUTPUT
// #define DEFAULT_I2C_SDA         gpio_num_t::GPIO_NUM_3
// #define DEFAULT_I2C_SCL         gpio_num_t::GPIO_NUM_17

// File Manager
#define SUPPORT_SD
#define SD_CARD_MISO_PIN        gpio_num_t::GPIO_NUM_4
#define SD_CARD_MOSI_PIN        gpio_num_t::GPIO_NUM_5
#define SD_CARD_CLK_PIN         gpio_num_t::GPIO_NUM_14
#define SD_CARD_CS_PIN          gpio_num_t::GPIO_NUM_15

Where is the PIN parameter changed (I.e. from input to output for example) Because MISO is input, but where the pin is told to function as input and not as output.

MartinMueller2003, 2024-11-27:
The usage controls the direction. That fact that a pin is MOSI will cause the SW to make that pin an output.

MartinMueller2003, 2024-11-27:
> There was an error in how lib_deps had git repo versions pinned. It was grabbing ArduinoJson 7.2.1 which breaks. I just pushed a fix to pin it to 7.2.0 for now - https://github.com/bblanchon/ArduinoJson/blob/7.x/CHANGELOG.md#breaking-changes

It has been working for me but my branch is 40 commits ahead of you. It was also compiling at the time the PR request was made.

forkineye, 2024-11-27:
> It has been working for me but my branch is 40 commits ahead of you. It was also compiling at the time the PR request was made.

Not sure what was going on. It started failing a while back in the pipeline and even after I'd merged your fork.

dutch-anykey, 2024-11-27:
okay,

success finally :) :) got the  sd card to work....more or less.

tried another SD-controller board (accepts micro sd cards directy instead of the other board that accepts mini SD) And switched from 3.3 volts (coming of the esp01 programmer) to 5 volts on the WT32 and on the card controller board (has a voltage regulator)
Now it is working, but perhaps the EN pin coming from the esp01 programmer is interfering as well, and I have to take a better look at various line that might need pull ups or pull downs (external)

br
Maurice

henrygab, 2024-11-27:
@dutch-anykey -- Note that WT32 has a history ... there are some fuse settings that may help you use the available pins more reliably.   This is due to how WT32 configured the ETH settings, and the ESP32's use of certain pins at boot.

Full details at [bootstrapping pins](https://gist.github.com/henrygab/653f5b4f76b00d4f71141a3d9db4abc1#file-960_bootstrapping_pins-md).

dutch-anykey, 2024-11-27:
Hi,

@henrygab 
Thanks for that info, will start reading this evening.
The more I play with this WT32 the more I understand that this device is really tricky indeed.

So I have it running now, but very unstable. It reboots quite often and SD card may be there or not...trying to make sense of it all. Is it the lab setup wiring, open connections on unused pins, wrong high/low status because of connection towards SD card reader....
Steep learning curve these last few days hehe.
Let's see if I can mange to set the fuse for at least io12, because that one i'm planning to use for sure. And 3.3 volt for flash is pretty much fixed for this device indeed.

henrygab, 2024-11-28:
@dutch-anykey 

#### Small form factor (~ D1 Mini size)

Yes, I had so many stability problems with the WT32 (even after setting the efuses), I stopped using it.  Reboots were common.  I think some folks found adding a large capacitor between a couple pins (EN and V?) gave them better stability, but I was fed up at that point.

I moved to using [Twilight Lord's shield](https://www.tindie.com/products/twilightlord/twilightlord-esp32-ethernet-shield/).  I was very happy with it, but it's now been out of stock for a couple years.  :(

Olimex has a [small PoE board](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware), but it has **_no galvanic isolation_**.   Danger!  I would not use for this reason.

I would **_definitely_** consider using the [wESP32](https://wesp32.com/).  (I couldn't find it in stock at the time I needed it.)  The wESP32 supports PoE, isolates the power, supports multiple voltages / PoE configurations, and just generally appears to be a solid board.  Currently priced ~$50, but at least it seems available via multiple channels now.

If ethernet is not a hard requirement, other options include 

#### If you're open to using a larger board....

I'm now using a QuinLed [Dig-Octa](https://quinled.info/quinled-dig-octa/); There's a distributor [for US/CA](https://www.drzzs.com/shop/dig-octa/), or direct [from CN](https://shop.allnetchina.cn/products/quinled-dig-octa-brainboard-32-8l).   This board has been rock-solid for me, and has a strong community on discord.

#### If ethernet is not a hard requirement....

ESPixelStick hardware is great and fit in 1.25" PVC pipe... I still have a number of these (with external antenna out one PVC cap, using power from the LED connection.   Very handy, just replace the top board with an appropriate ESP32 board.

QuinLed also has [a number of boards](https://quinled.info/addressable-digital-leds/#DigitalBoards)... check them out.  Many varieties and form factors.  Might be something you could use.

dutch-anykey, 2024-11-28:
@henrygab 

Thank you for the information. I was also already looking for alternatives and the quinled products are really nice. It basically covers what I wanted to achieve.
It's not for an Xmas display, but we have a yearly parade (carnaval) where my friends and me build a parade wagon with lights. 
I already have a falcon f16v3 and a kulp k40 pb differential controller + 10 differential receivers. But as it goes....each year more outputs needed hehe. So I was looking for a cheaper alternative to expand the amount of ports, and also this year the plan is to have mobile led suits (pebble/seed pixels) that are also synced into the show.... ambitions are set high...

So the "normal" esp32's are planned for mobile suits via wifi, but plan is to have them equipped with a SD card and controlled via  FPP remote to sync with the main show. That's why went for espixelstick software. And so far this solution seems to be running stable. esp32+SD+4 outputs. Not sure when I have finished 10 of those how the sync and wifi network will hold up. But at least I could have them play a local fseq from the SD if connection is lost. So the espixelstick software covers all my needs, also the integration in Xlights for uploading sequence files and upload output configuration is brilliant. (lot's of things that wled doesn't have) 

And for expansion of the main show ports I planned to have these WT32 boards. Ethernet+ 4 outputs and SD card to run them also in FPP remote. But maybe for that SD is not needed or overkill perhaps? Maybe 100 Mbits/s and the WT32 performance it will keep up with normal DDP traffic? What is your opinion? Or is the instability also there without the SD config that you know?
Or would a normal ESP32 with external LAN module also be an option?

I'm already in to deep I guess. I bought lot's of stuff and components already (like sn74ahct125 ic's for level shifter, dc/dc buck convertors etc..) + I like to build things. 
But now it seems that the WT32 will be a blocking problem. It is a hobby project (also the parade wagon) So I don't expect it all to be professional quality. But on the other hand .... I don't want to be the Griswold guy from National Lampoon's Christmas vacation fiddling with his lights all day hahaha.

I did spent a little time again this evening on the WT32, it seems to run okay as long as I don't try to write something to the SD card, then it will reboot. I placed the elco between EN and GND for boot delay and managed to do the fuse burn for the 3.3 voltage setting. But it doesn't make it better. Maybe leave it for a few days and start fresh again.

br
Maurice

henrygab, 2024-11-29:
> And for expansion of the main show ports I planned to have these WT32 boards. Ethernet+ 4 outputs and SD card to run them also in FPP remote.

Thanks for the background ... With F16v3 and Kulp controllers, you're well past the starting point with this hobby....

If I understand your design:
* FPP files on all controllers, accessible via SD card
* xlights (or similar) for uploading the FPP files, and initiating playback on all controllers
* Main parade wagon: controller(s) with ethernet + SD card w/ .fseq files + FPP remote supported
* Mobile suits: controller with WiFi + SD card w/ .fseq files + FPP remote supported

That seems a very solid design.  Presuming that the WiFi signal isn't disrupted, I understand that `WiFi + SD w/ .fseq files + FPP Remote` should be very effective, as essentially the .fseq files can be sent to each controller ahead of time, so that the only thing WiFi is needed for during the show is to synchronize the playback between the player nodes.   However, I have not yet used this configuration myself ... so I can't speak to any "gotcha!" / rough edges.

<hr/>
<details><summary>I gave up on WT32 boards...</summary><p/>

In my experience, when I tried to use WT32 boards, I was unable to achieve stability.  One question you'll have to answer, based on your circumstances, is how much your time is worth vs. risk of hardware issues that are hard to track down.  Is the WT32 issue caused by an undersized power regulator?  Is it caused by insufficient power caps?  Is it just much more sensitive to power supply noise?   I never figured out the root cause ... I ended up throwing my WT32 boards away, because it wasn't worth the headache for me ... and as you can see in the gist, I did perform some investigations before I hit that point;  It just wasn't something I could rely upon.

For me, it was worth the price to pay more per board, and have a more stable hardware base to build on ... as I also wanted to stop being Griswold.

</details><hr/>

dutch-anykey, 2024-11-29:
> with F16v3 and Kulp controllers, you're well past the starting point with this hobby.

yeah, I think I have the falcon already for 4 or 5 years and the Kulp followed a few years later with buying remote receivers along the way each year. I really had to convince my friends that we needed those. (And they were cheaper a few years ago.)
But so far they performed rock solid.

Being the only led-light enthusiast /electrician in our group that sometimes is a challenge. And then when you have the gadgets the other problems start like planning/programming, flickering leds, power injections I'm sure you know what I'm talking about :)
At times it starts to feel like self-flagellation to get everything going hahaha.
Last year I build matrixes from seed pixels (5 cm  distance between pixels) They liked that, so I opted for this year to have 2,5 cm distance between them to have more resolution....but you need 4 times more seed pixels to get the same size matrix for that and therefor lot's more physical outputs.

>I gave up on WT32 boards...

I think I'm going to call it quits as well on this wt32. The simple plan was to flash the wt32 with pixelstick, stick a SD reader, dc/dc converter and level convertors to it and start using it. How little did I know ... haha. 
Even looking at pictures of pcb boards from mottramlabs, wasatchpixels and some others to see if some extra components are there  perhaps. But so far pretty basic setups. That makes me wonder if it is stable or how they made it stable. 
So as a last ... I could build and test the wasatch built that is in the espixel repositery and skip the SD card for that and use it for live DDP only.

I learned a lot the last few days, that's always nice :)

MartinMueller2003, 2024-11-29:
As I was adding platforms, two platforms went to the list of "Never, EVER, use these". WT32 and ESP32 CAM boards. Both had major problems in the brownout circuits and general issues with stability.

dutch-anykey, 2024-12-01:
I'm still fumbling with the wt32 - no sd card. (can't let it go, while I should probably) Looks more stable.
But now I have problems with xlights + espixelstick. 
Upload of controller data is working. But after the first time when I go back in Xlights > layout And there look at a model > controller section. Some options disappeared.  I can only see port, protocol, snd set zigzag. Other settings like color order, brightness and another few are gone. Is that something that is known? Or should it work? Not sure if this is espixelstick related or Xlights. Else I can create a report with more details.

Just to share what we are doing with all of this :)
It's of last year. just a test seq running
https://www.youtube.com/watch?v=16m4KDNcOOc

kind regard,
Maurice

dutch-anykey, 2024-12-04:
@MartinMueller2003 

I opened a ticket @xlights two days ago. Let's wait and see.

Got lucky, my friends gave a green light to order a kulp controller and remote receivers. That's a goodbye to the wt32 project for me and less stress to get that going. 

Now busy with the ESP32 (generic?) boards. Test is doing well. It's running for more than a day now with SD and 4 outputs. That's a nice change to see compared to the wt32. 
I do still have question marks with building the sw. Do I still need to build and upload the filesystem myself in VScode?
Because I have some weird behavior. Also erasing flash says it completed successfully, but still old setting on ports a wifi are still there even after uploading a new build (there is noting in the platformio_user.ini file)

MartinMueller2003, 2024-12-04:
Yes you need to upload the filesystem yourself using VSCODE. That is one way to reset the configuration. another is to use the factory reset on the admin page.