forkineye/ESPixelStick #799: Weird Flickering

stijn-dhondt, 2024-12-03:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
4.0-ci11980630208

**Hardware Version**
Dig quad v3 with ethernet

**Binary release or compiled yourself?**

**Operating System (and version)**

**Web Browser (and version)**

**Access Point**
Ubiquity

**Describe the bug**
When running wled i notice no flickering when running local effects or trough DDP
When running the firmware mentioned i have flickering on local effects and ddp data even local sequence.

Tested on 2 dig quad's
Tested with 2 similar and one different meanwell power supply
Tested with wireless and ethernet on both
Replaced all cabeling on one controller
Wled on both controllers with all 3 powersupply's generete no issues at all.

MartinMueller2003, 2024-12-03:
I have been chasing this for a while. The experimental version no longer has a flicker on the built in effects but still has some on the fseq playback.

stijn-dhondt, 2024-12-04:
Hi @Martin,

For what i found even the local effects had the issue...
I have the same release running on Dig Octa's with no issues. That's why i have been testing like hell with all kinds of powersupply's , cabeling and even 3 different esp's on the dig Quad boards just to be sure.

On all tree with wled no issues for now. but would like to use ESPixelStick on all of them...

Thx for investigating!

onewithhammer, 2024-12-04:
I flashed the latest version yesterday for DIG-QUAD with AE+ and the flicker is still there.   Primary input = disabled.  Secondary input = effects.  Effect = solid or rainbow.  FTP = disabled.  No browser connected.  Dedicated WIFI show network.

onewithhammer, 2024-12-04:
I flashed latest version yesterday just now on a Tetra2go - same setup..pixels, PSU, network...No flicker running local effects.

MartinMueller2003, 2024-12-07:
Beginning to think this is a HW issue.  I have modified the pixel timing a bit to see if the HW works better.

MartinMueller2003, 2024-12-08:
With current timing, I see a flash about once an hour.

stijn-dhondt, 2024-12-08:
will test tomorrow, fingers crossed ;-) Thx Martin!

Terrevue, 2024-12-12:
I am experiencing the flickering as well on multiple ESP32 (WROOM) based devices.  I am using a logic level shifter on both and have tried every build available since Sept 2024 with the same results.  Using the same device, I installed WLED and there were zero flickering issues.  

On a side note (which is somewhat applicable to this thread, since I can't get any of the 2024 builds to work correctly), is the beta 5 version from 2023 a viable alternative?  Last year there were issues with the number of files on the SD card.  Was that corrected in beta 5?  Is that considered the last stable version?

MartinMueller2003, 2024-12-28:
I found an issue in which the WiFi and Ethernet interrupts caused a significant delay in the ability to service the RMT interrupts, This caused a transmitter underrun and misalignment in the output. AKA Flashing random stuff. This is in the current PR.

closed 2024-12-28 by MartinMueller2003