forkineye/ESPixelStick #802: Corrupt disk

cybercop23, 2024-12-06:
Running either Aug 17, Sep 23 or Oct 21 versions of v4 and uploading to an SD card from xLights, when I tried to delete a file from the GUI File Management, it corrupted the filesystem and got the ESP into a continous boot. Once I replaced the card it was fine. This happened twice on 2 diff HW and 2 diff versions. Using Chrome Version 131.0.6778.109 (Official Build) (64-bit)

MartinMueller2003, 2024-12-06:
This is part of the same issue that is causing the xLights updates to fail. I had to slow things down a bit and I am getting a bit more reliability in the updates. I should have the changes added to the experimental build soon.

MartinMueller2003, 2024-12-06:
Webflasher has been updated with a new experimental version that has increased (but not perfect) transfer reliability and fewer SD card hangs. FYI: For me, putting the SD card into a PC and formatting tends to correct the issues.

cybercop23, 2024-12-06:
> FYI: For me, putting the SD card into a PC and formatting tends to correct the issues.

Right.. that's how I got it to come back to life... well.. first replaced the card and then it worked so I thought I had 2 bad cards and that seemed odd and then reformatted. I'll try it today. Thanks much.

joe8012, 2024-12-06:
chkdsk /f also works if you don't want to re-upload all your sequences again.  After many uploads over the past year, I found that having the espixelstick UI interface closed during the fpp connect upload, greatly reduces the upload failures.

cybercop23, 2024-12-06:
Yeah, same here, and that's what I did. But wanted to delete a file manaually and test some changes from xLights and then it went poof.

cybercop23, 2024-12-06:
@MartinMueller2003 I don't see the new ver? Refreshed and still nada...
[image]

dutch-anykey, 2024-12-07:
I had that too.
But you need to select this I found out:
[image]

It doesn't show any version there, but after I flashed it I got:

FW Version4.0-ci12088826635
Build DateNov 29 2024 - 17:46:35

MartinMueller2003, 2024-12-07:
The CI builds have been updated with some additional changes to prevent corruption of the SD card. PR has been generated.

cybercop23, 2024-12-08:
Thank you very much.

dutch-anykey, 2024-12-10:
@MartinMueller2003 

I'm testing with SD-card and Xlights.
But still can't get it stable, bust guess from my side that it is related to this ticket.

When I startup SD is present, even a manual  upload of a big seq. file succeeded (took a while as known)

Also I did a controller config upload from Xlights (still not working correctly and @Xlights github so far not responded to my question) 
(Could it be that this upload mangles something in the ESPixelstick?)

But anyway the SD card stopped working also a few reboots didn't help
I attached the logfile with the  events. A few extra reboots are there because wifi sometimes doesn't connect immediately 

[espixelstick_log.txt](https://github.com/user-attachments/files/18084662/espixelstick_log.txt)

MartinMueller2003, 2024-12-10:
There is an issue in the SD code that sometimes causes the SD to lockup. You need to turn off power (IE reseat) the SD. I am still working to find a way to avoid that state. On my other projects I added a power control for the SD card so it can be hard reset.

cybercop23, 2024-12-11:
Like a "Lock" button to move all ports brightnesses together

MartinMueller2003, 2025-02-18:
Have not seen this in the current builds.

closed 2025-02-18 by MartinMueller2003