forkineye/ESPixelStick #805: Packet errors count is the same as total packets, is that accurate?

patdelaney, 2024-12-08:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
FW Version4.0-ci12216055840
Build DateDec 7 2024 - 21:01:59

**Hardware Version**
d1_mini_pro

**Binary release or compiled yourself?**
Web updater Experimental 

**Web Browser (and version)**
Chrome  Version 131.0.6778.109 (Official Build) (64-bit)

**Describe the bug**
When I view the status page of pixelstick while sending e1.31 data from da_E131 tester, the packet errors is increasing with each packet that comes in. I never noticed that before, and I'm not sure if its a bug.

When I use xlights in test mode the same thing happens.
[image]

MartinMueller2003, 2024-12-08:
I thought I had that fixed. The problem is that some data sources always send a zero in the sequence number field. The code expects the next number to be a one, gets a zero and reports an error. I had at one point added logic to not increment the error count when the source sequence number is zero. I will check to see why that is not working.

MartinMueller2003, 2024-12-08:
Found the path where it did not have the zero check in place. This is now in the latest version on the web flasher and is in a PR.

KHouse75, 2024-12-10:
I think this may still be an issue.  I have 4.0-ci12223533891 installed and packet errors are increasing along with total packet but everything is working as expected.

[image]

MartinMueller2003, 2024-12-19:
Found another issue in this code and uploaded another fix. This will be in the first PR in January.

MartinMueller2003, 2025-02-18:
This fix is on main.

closed 2025-02-18 by MartinMueller2003