forkineye/ESPixelStick #806: Wifi does not connect to strongest AP

mdforbis, 2024-12-08:
--------- Instructions --------
Please provide answers directly below each section.
--------- Instructions ---------

**ESPixelStick Firmware Version**
[ESPixelStick v4.0-beta5](https://github.com/forkineye/ESPixelStick/releases/tag/v4.0-beta5)

**Hardware Version**
QuinLED Dig Uno homebuilt with ESP32

**Binary release or compiled yourself?**
binary

**Operating System (and version)**

**Web Browser (and version)**

**Access Point**
3 TP Link AP's

**Describe the bug**
I have three AP's.  One in the yard for the lights, one in the garage (mostly for the house) and one in the basement (to cover the other half of the house).  When the firmware searches for an AP, it seems to use the first one it finds (i.e, connects to the one on Channel 1) even though this is the weakest one based on it's location.  Would be nice to have it search all channels and choose the strongest signal before connecting to the SSID.  When it does connect to the weakest one, the data rate is so slow that you can't connect to the web interface most of the time.

(Obviously can work around it by using separate SSID's per AP, but this means it's more difficult to take from production in the yard to the basement to work on a problem without reflashing.

*Could also be considered an enhancement request over a bug, was undecided.

KHouse75, 2024-12-09:
Same issue here.  I have 4 mesh eero devices and if I connect to one inside my house while troubleshooting then move to the permanent location outside the house, it tries to connect to the access point it was connected to originally.  The SSID is the same but WiFi used by ESPixelStick remembers the individual access point instead of connecting to the strongest one.

MartinMueller2003, 2024-12-09:
> Same issue here. I have 4 mesh eero devices and if I connect to one inside my house while troubleshooting then move to the permanent location outside the house, it tries to connect to the access [...]

This is incorrect. There is nothing in the WiFi layer that records the BSSID of an AP and uses that as a preferential AP. At the start of the connect sequence, the ESP waits for the first SSID broadcast it receives and connects to that. First come, first served. There is also no code that looks for the "Best" BSSID and uses that as the desired AP. That would be an enhancement.

KHouse75, 2024-12-09:
> This is incorrect. There is nothing in the WiFi layer that records the BSSID of an AP and uses that as a preferential AP. At the start of the connect sequence, the ESP waits for the first SSID [...]

I had looked over the code and didn't see anything referencing bssid.  What's interesting is if I freshly flash the device, it'll connect to the AP 5 ft from me instantly.  If I then take it outside where the other AP is located, far outside the range of the original AP it connected to and standing 5 ft from the outside AP, it will not connect to the outside AP initially.  If I leave it for several minutes (maybe 10+), it'll eventually connect.  If I bring it back inside a few ft from the original AP, it'll connect instantly.

If I flash it near the outside AP, it'll connect instantly to the outside AP.  If I bring it inside near the inside AP, it won't connect initially.  It'll again take several minutes before it connects to the inside AP.

I've also tested this in a lab using 2 APs in a mesh by physically unplugging power to the AP I had first connected to.  It takes several minutes to connect to the second AP but will connect instantly if I plug in the first AP.

I've experienced this with ESPHome devices as well and debugging showed that it was indeed trying to connect to the original AP BSSID.

Additional research suggests that the WiFi info including BSSID is stored in EEPROM on the ESP8266 and is used outside of the actual code on boot.  I don't know if this is specific to a particular code or library that's used.  I have not researched this any further.

MartinMueller2003, 2024-12-09:
FYI: BSSID is the mac address of the wireless interface on the AP. It is used to direct messages to a specific AP (just a bit of trivia). IMO It is a poor choice of name but that is what is in the spec (It might actually be BSID but I have not checked in over a decade).

KHouse75, 2024-12-09:
Correct.  You can force connection to a particular SSID/BSSID combo which should resolve such an issue but coding it to determine which AP advertising the same SSID is the strongest/closest would be preferred, if possible.

davron12, 2024-12-13:
In my experience, and I have 5 APs in use in my display, it eventually connects to the strongest one or at least one with a good signal. Also, "strongest" signal could also be constantly changing due to environmental factors, and you wouldn't want it constantly switching. I see signal strength variations almost up to 6-10dBi as the show is running. As long as it has a strong signal, it doesn't need to change.