forkineye/ESPixelStick #809: factory reset not working

dutch-anykey, 2024-12-09:
Factory reset is not deleting everything.

Wifi settings are deleted, but all settings in device settings remain
Tested on two boards:

FW Version 4.0-ci12223533891
Build Date Dec 8 2024 - 16:45:29
Architecture ESP32
Board Name d1_mini32  <**<actually selected generic esp32 3 mb via web flasher. But shows d1_mini32**
Flash Size 4194304
Flash ID84158F0A8AB4

and

FW Version 4.0-ci12223533891
Build Date Dec 8 2024 - 16:47:59
Architecture ESP32
Board Name esp32_wasatch **<< this is a wt32 board**
Flash Size8388608
Flash IDE4C12D65B7A0

br Maurice

MartinMueller2003, 2024-12-09:
This is an odd issue. The reset DID work. I just did it and verified all settings are gone. The problem is that the Browser still had all of the old settings in cache. I had to tell the browser to clear its internal caches before the correct information was displayed.

dutch-anykey, 2024-12-09:
confirmed here as well,
After clearing the cache (chrome browser) All shows correct.

MartinMueller2003, 2024-12-09:
Will be on the Web Flasher in a few hours. Simply added a mechanism on the UI to cause the next config to use a different url than the previous request. That causes the cache to get bypassed.

dutch-anykey, 2024-12-10:
I checked and this is working now. Thank you

closed 2024-12-10 by MartinMueller2003